realsenseai/librealsense #525: RGBD stream stopping after 1hour with increasing ram requirement

arpit15, 2017-06-08:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | ubuntu 14.04                |
| Kernel                | 4.4.0-66-generic                |
| ROS                   | Indigo                  |
| ROS RealSense         | 1.8.0               |
| librealsense          | 1.12.1               |
| Camera Type-Firmware  | SR300 3.15.0.0            |

---

### Expected Behavior
The depth stream from topic /camera/depth_registered/points is obtained at all times. 

### Actual Behavior
The ram requirements of the process keeps increasing until it reaches to 95% of the total ram which in my case is 32GB. After eating up the ram the sensor stops publishing the topic. 
I have seen following messages after the node is shutdown

1. rs_wait_for_frames
2. [camera/camera_nodelet_manager-1] process has died [pid 10166, exit code -9, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager 
3. Without any message
I could receive more 100k frames in rviz before the node crashes.
I can't understand why the requirement of memory keeps increasing as the nodes don't store anything while processing input data. 
Kindly let me know what could be a way to run the sensor for longer durations.

### Steps to Reproduce
roslaunch realsense_camera sr300_nodelet_rgbd.launch

Thank you
Arpit

oordan, 2017-08-09:
Related to [intel-ros/realsense/issues/243](https://github.com/intel-ros/realsense/issues/243)

jlgarcia75, 2017-11-01:
Hello @arpit15,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

arpit15, 2017-11-02:
@jlgarcia75 I was facing the same problem a month ago. I have not being using the sensor due to mis-alignment issues in depth and rgb images. However there are no updates in librealsense, should I hope any different results?

jlgarcia75, 2017-11-08:
Hello @arpit15,

The misalignment issues can be due to a miscalibrated camera which is unfortunate in an SR300 because it cannot be calibrated in the field.

Have you tried to reproduce with the latest version of RealSense ROS and Librealsense?

This is the ROS repository that we support: https://github.com/intel-ros/realsense/releases
This is TAG 2.0.1 of the repository, so it is the latest release. 
The Release Notes of this release states that it supports librealsense 2.8.1.

Regards,
Jesus G.
Intel Customer Support

arpit15, 2017-11-08:
@jlgarcia75 I have not tried with TAG 2.0.1. I will be able to check the issue in the next week and post the results.

jlgarcia75, 2017-12-21:
Hello @arpit15,

Do you still need help with this issue.

Regards,
Jesus 
Intel Customer Support

dorodnic, 2018-01-10:
I am closing this issue since it never reproduced on any of our systems and for the best of our knowledge for no other customer. 
Please update if you find the root cause for the leak on your setup.

closed 2018-01-10 by dorodnic