realsenseai/librealsense #527: multi-SR300 save color sequences issue

dan0910, 2017-06-09:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Win10 Pro |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013 update5| 

Hi，
I need  use **two** realsense SR300 to shoot the same scene, and record both color and depth image sequences.
While opening the depth stream of each camera at the same time, infrared light will interfere with each other.
So I need control two realsense turn on and off alternately by programming.
Here is my logical flow chart:
[image]

After running my code, I find that the depth image sequences of both cameras are intact(That's good), but there are always several frames turning black(no data?) randomly in both color image sequences, just like this:

[image]

When running the code,the console has the following error message:

> rs.error: ReadSample returned MF_E_NOTACCEPTING
> rs.error: ReadSample returned MF_E_NOTACCEPTING
> rs.error: ReadSample returned MF_E_NOTACCEPTING
> ...........

I want to get completed color and depth image sequences of both cameras.

Here is my code, could you please help me to solve the problem? Thank you very much!

```
#include <librealsense/rs.hpp>
#include "example.hpp"
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/imgproc.hpp>
#include <iostream>
#include <algorithm>
#include <Windows.h>
using namespace cv;
using namespace std;
std::vector<texture_buffer> buffers;
void main() 
{
	//frame index
	int frameFlag = 0;
	int frameFlag2 = 0;

	rs::log_to_console(rs::log_severity::warn);
	rs::context ctx;
	if (ctx.get_device_count() == 0) throw std::runtime_error("No device detected. Is it plugged in?");

   // get two devices
	rs::device & dev1 = *ctx.get_device(0);
	rs::device & dev2 = *ctx.get_device(1);

	//cam flag
	int camFlag = 1;
	uchar *pRgb = NULL;
	while (true)
	{
		if (camFlag == 1)
		{
                        //Turn on camera ①
			dev1.enable_stream(rs::stream::depth, rs::preset::best_quality);
			dev1.enable_stream(rs::stream::color, rs::preset::best_quality);
			dev1.start();

			dev1.wait_for_frames();		
			pRgb = (uchar*)dev1.get_frame_data(rs::stream::color);
			uint16_t *depthImage = (uint16_t *)dev1.get_frame_data(rs::stream::depth);
			uchar* pD2C = (uchar*)dev1.get_frame_data(rs::stream::depth_aligned_to_color);

			char fileName[100];
			Mat rgb_show;
			Mat rgb(480, 640, CV_8UC3, pRgb);
			cvtColor(rgb, rgb_show, CV_BGR2RGB);
			imshow("Realsense-RGBImage", rgb_show);
			sprintf_s(fileName, "F:\\realsense\\NET\\color\\%04d.png", frameFlag);
			imwrite(fileName, rgb_show);

			Mat depth(480, 640, CV_16UC1, depthImage);
			imshow("Realsense-Depthimage", depth * 15);
			Mat D2C_show;
			Mat D2C(480, 640, CV_16UC1, pD2C);
			imshow("Realsense-D2C", D2C * 10);			
			sprintf_s(fileName, "F:\\realsense\\NET\\depth2C\\%04d.png", frameFlag);
			imwrite(fileName, D2C);

			frameFlag++;
			waitKey(1000);

			//turn off camera①
			dev1.stop();
			dev1.disable_stream(rs::stream::depth);
			dev1.disable_stream(rs::stream::color);			
			//camflag setted to2
			camFlag = 2;
		}
		else
		{
			//turn on camera②
			dev2.enable_stream(rs::stream::depth, rs::preset::best_quality);
			dev2.enable_stream(rs::stream::color, rs::preset::best_quality);
			dev2.start();

			//save images
			dev2.wait_for_frames();
			/*uchar**/ pRgb = (uchar*)dev2.get_frame_data(rs::stream::color);
			uint16_t *depthImage = (uint16_t *)dev2.get_frame_data(rs::stream::depth);
			uchar* pD2C = (uchar*)dev2.get_frame_data(rs::stream::depth_aligned_to_color);

			char fileName[100];
			Mat rgb_show;
			Mat rgb(480, 640, CV_8UC3, pRgb);
			cvtColor(rgb, rgb_show, CV_BGR2RGB);
			imshow("Realsense-RGBImage", rgb_show);
			sprintf_s(fileName, "F:\\realsense\\NET2\\color\\%04d.png", frameFlag2);
			imwrite(fileName, rgb_show);

			Mat depth(480, 640, CV_16UC1, depthImage);
			imshow("Realsense-Depthimage", depth * 15);
			Mat D2C_show;
			Mat D2C(480, 640, CV_16UC1, pD2C);
			imshow("Realsense-D2C", D2C * 10);
			sprintf_s(fileName, "F:\\realsense\\NET2\\depth2C\\%04d.png", frameFlag2);
			imwrite(fileName, D2C);

			frameFlag2 ++;			
			waitKey(1000);		

			//turn off camera②
			dev2.stop();
			dev2.disable_stream(rs::stream::depth);
			dev2.disable_stream(rs::stream::color);
			//camflag setted to 1
			camFlag = 1;			
		}
	}
}
```

dorodnic, 2017-06-11:
What happens if you move `context` creation (and device selection) into the relevant scopes? 
Something like:
```cpp
while (true)
{
	if (camFlag == 1)
	{
		rs::context ctx;
		rs::device & dev1 = *ctx.get_device(0);
		...
	}
	else
	{
		rs::context ctx;
		rs::device & dev1 = *ctx.get_device(1);
		...
	}
}
```
I propose this because lifetime of the underlying Media-Foundation objects is tied to `context` lifetime.

dan0910, 2017-06-12:
@dorodnic 
Hi~
I follow your suggestion to move them into the relevant scopes.
But it comes an Unhandled exception
[image]
 at `dev1.start();`
```
class error : public std::runtime_error
{... ...
        static void handle(rs_error * e) { if(e) throw error(e); }
```
[image]

By the way, I used  debug mode compiling before, but there had always been a run-time error about msvcr120.dll. So I use release mode compiling to avoid the error. But now it comes again. ╮(╯-╰)╭

delmottea, 2017-06-13:
You don't need to close the stream, you can set the laser power to 0.
Check https://github.com/IntelRealSense/librealsense/issues/482
The effect is not instantaneous so you need to wait a few frames before the laser turn on or off (my tests shows that waiting 3 frames (at 10 fps) was enough).
Take care that you could get this issue too : https://github.com/IntelRealSense/librealsense/issues/519 (only got it one time, restarting the computer (not just reset usb) solved the issue)

dan0910, 2017-06-14:
@delmottea 
Hi，thank you very much!
It works!

closed 2017-06-14 by dan0910