realsenseai/librealsense #528: SR300 Crash While Capturing

nizihabi, 2017-06-13:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version |   | 
| Operating System & Version | Win10 15063  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013  | 

The application could crash down and print "ReadSample returned MF_E_VIDEO_RECORDING_DEVICE_INVALIDATED"

The camera is using  200fps and ir stream

oordan, 2017-07-27:
This error usually indicates camera disconnect. Currently, librealsense is not capable of handling dynamic disconnections, this is something that will be resolved in future versions. 
To try to troubleshoot why the camera might disconnect, please contact [1x1 support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html)

jlgarcia75, 2017-10-03:
Hello @nizihabi,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75