realsenseai/librealsense #529: Once stopped, cannot find the camera again

nlyubova, 2017-06-13:
| Required Info  |   |
|---|---|
| Camera Model | all of them: R200, F200 or SR300 | 
| Firmware Version |  2.0.71.28 | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.4.15-tegra on Jetson TX2 |
| Build System |  Makefile | 

Can start the driver only once and then it does not detect the camera anymore and needs to  disconnect the camera from USB and connect it again to make it working. Tested with ZR300, R200, SR300, all cameras show the same behavior. Does it mean that the camera driver does not stop correctly?

Please help!

delmottea, 2017-06-15:
I have the same problem on my computer (but not all the time, and mainly when used in multiple camera configuration).
Someone from my team solved the problem by creating a script to reset the usb bus before starting the program. If you want avoid to reset the whole bus, it's also possible to simulate a usb disconnect-reconnect by getting the bus and device id (lsusb -d 8086:0a80 for R200, 8086:0aa5 for SR300, 041e:409b when the camera is wrongly recognized as a Creative camera), and using this program to do the reconnection : https://gist.github.com/x2q/5124616
and using it like sudo ./usbreset /dev/bus/usb/002/007 (replace 002 by your bus id and 007 by your device id)

JeremyHX, 2017-07-03:
Same issue, but the usbreset doesn't help in my case.
It also happens every single time.

After fresh boot:
```
nvidia@jetson:~$ ./librealsense/build/examples/cpp-headless
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense ZR300
    Serial number: 3511804088
    Firmware version: 2.0.71.28
Writing cpp-headless-output-DEPTH.png, 480 x 360 pixels
Writing cpp-headless-output-COLOR.png, 640 x 480 pixels
Writing cpp-headless-output-INFRARED.png, 480 x 360 pixels
Writing cpp-headless-output-INFRARED2.png, 480 x 360 pixels
Writing cpp-headless-output-FISHEYE.png, 640 x 480 pixels
wrote frames to current working directory.
nvidia@jetson:~$ ./librealsense/build/examples/cpp-headless
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense ZR300
    Serial number: 3511804088
    Firmware version: 2.0.71.28
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
RealSense error calling rs_wait_for_frames(device:0x464ef0):
    Timeout waiting for frames.
nvidia@jetson:~$ ./librealsense/build/examples/cpp-headless
RealSense error calling rs_create_context(api_version:11201):
    UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error
```

oordan, 2017-07-27:
related to [intel-ros/realsense/issues/247](https://github.com/intel-ros/realsense/issues/247)

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75