realsenseai/librealsense #537: 【ZR300】Camera & IMU intrinsics and Extrinsics unreliable from API?

sqn175, 2017-06-17:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | 2.0.71.28  | 
| Operating System & Version | ubuntu 14.04 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-75-generic |

Hi,

I have a realsense zr300 camera used for vision-inertial odometry, in which case the extrinsics matrix from color camera to motion module is needed.
So how can I get the extrinsics ?
 
I have tried the function
`dev->get_motion_extrinsics_from(rs::stream::color)`

provided by librealsense API. Well the function returns error, I look into the function and find that the error comes from
`throw std::runtime_error("Motion intrinsic is not valid");`

But I can get the motion intrinsics using the function
`dev->get_motion_intrinsics();`

So how to solve the problem? Can I get the extrinsics directly from the library? If not, should I calibrate the extrinsics myself?

Btw, is the camera&IMU intrinsics results derived from the API reliable?
```
    colorIntrinsic = dev->get_stream_intrinsics(rs::stream::color);
    imuIntrinsic = dev->get_motion_intrinsics();
```

Thanks in advance.

dorodnic, 2017-06-17:
Hi @sqn175 
What about `dev->get_motion_extrinsics_from(rs::stream::depth)` and `dev->get_motion_extrinsics_from(rs::stream::fisheye)`?

sqn175, 2017-06-17:
Hi, @dorodnic ,

`dev->get_motion_extrinsics_from(rs::stream::fisheye)` works and returns a reasonable result:

        [ 0.999864817, 0.0162242502, 0.00268337899, 0.0068119932,
          -0.0161926001, 0.999803603, -0.0114245601, 0.00240068603,
          -0.00286820694, 0.0113795698, 0.999931097, -7.7811e-06,
          0.0, 0.0, 0.0, 1.0],

But I can not find a way to validate the result.

dorodnic, 2017-06-18:
I suspect this to be a bug in the calibration process. It seems like MM-Color and MM-Depth tables are empty. As a work-around, I believe you can get the relevant extrinsics by multiplying extrinsics from Depth to FE by extrinsics from FE to MM (this multiplication might cause some small degradation in precision, that's why librealsense tries to read from dedicated tables)

closed 2017-07-27 by oordan