realsenseai/librealsense #538: How to us rs_motion_device_intrinsic confirm

hlc111, 2017-06-20:
| Required Info  |  How to us rs_motion_device_intrinsic confirm |
| Camera Model |  ZR300 | 
Gyro intrinsic like below:
    /* Scale X        cross axis        cross axis      Bias X */
    /* cross axis     Scale Y           cross axis      Bias Y */
    /* cross axis     cross axis        Scale Z         Bias Z */
and i confirmed all cross  axis value is 0.
So real value should be (raw -Bias)*Scale  (should not be  (raw +Bias)*Scale ).
e.g. GyroX = (RawGyroX -Bias X )*Scale X.
Am i right?

Thanks!

dorodnic, 2017-07-13:
Hi @hlc111,
It should be GyroX = (RawGyroX * Scale X) - Bias X.
If you apply the same formula to the accelerometer values, you should get sqrt(x^2 + y^2 + z^2) = 9.8 when the camera is in inertial reference frame.

closed 2017-07-13 by dorodnic

cza1019, 2017-08-11:
@dorodnic 
What does " when the camera is in inertial reference frame" mean? I do the same formula to the accelerometer values, but the mean of sqrt(x^2 + y^2 + z^2) isn't equal to 9.8.

dorodnic, 2017-08-20:
The camera isn't moving. Can you share more info about the device? (the output of enumerate-devices)