realsenseai/librealsense #539: ZR300 on Nvidia Jetson TX2 not reporting IMU data

cvilas, 2017-06-21:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | 2.0.71.28  | 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  |  4.4.15 aarch64 |

Following [these instructions on Jetsonhacks](http://www.jetsonhacks.com/2017/03/26/intel-realsense-camera-installation-nvidia-jetson-tx2/) I am able to get the ZR300 to stream all camera images on the Jetson TX2. However, on running `cpp-motion-module` I don't see the motion callback ever being triggered. Any insights on why this might be? The feature of most interest to me on the ZR300 is its ability to deliver Images synchronised with the IMU.

dorodnic, 2017-08-28:
Hi @cvilas 
Just wanted to update that we reproduce this problem and trying to look into it. Please take into account that the ZR300 does not officially support Jetson TX2. We will do our best to understand the problem and offer a work-around.

cvilas, 2017-08-28:
Thanks very much @dorodnic. I appreciate your time. Of course, I understand that the TX2 is not one of the supported platforms, but it is an important one in my opinion. Especially as Intel's own Joule boards are not supported anymore.

closed 2017-10-17 by jlgarcia75

seath49, 2017-10-18:
Hello everyone, 

I tried ZR300 + Jetson TX1 with the same result - no motion data coming out. I didn't post anything here, since I didn't have enough time to get into more debugging. I see this issue being close, however, unresolved. Is there a chance for a solution in the future?

Thanks, 
Seath

yc718764666, 2018-05-23:
i get the same problem,does anybody can give me a suggestion?

maxwu8416, 2019-06-26:
hi~so is there anybody already solved this problem !?
i also have this problem and stop right here around a week....
hope someone can share your solution...

seath49, 2019-06-26:
Hi, it's been a long time, not sure if I recollect this correctly ... 
the issue was with the carrier board of TX2, not supporting usb3.
The manufacturer confessed at the end.
I'll try to reach out to my colleague who suffered this just recently.

maxwu8416, 2019-06-26:
@seath49  thamk you!! 
 i really appreciated that your reply, and also bothering you to reach out your colleague, if there has any new information please let me know !
Thanks again, Expect your update