realsenseai/librealsense #542: Hi, I am trying to calculate xyz coordinate from a pixel of rgb image using the formula "X= zx/f" instead of projection function in rs-sdk. However the offset of the x coordinate that I get directly from rs-sdk and the formula(X= zx/f) is not a const value. Could you please tell me the reason.

qwzlh, 2017-06-26:
Hi, I am trying to calculate xyz coordinate from a pixel of rgb image using the formula "X= zx/f" instead of projection function in rs-sdk. However the offset of the x coordinate that I get directly from rs-sdk and the formula(X= zx/f) is not a const value. Could you please tell me the reason.

dorodnic, 2017-07-01:
Have you tried using `rs_deproject_pixel_to_point`? SR300 color stream contains a distortion that you need to correct based on calibration parameters (stream intrinsics)

qwzlh, 2017-07-01:
hi @dorodnic 
How can i get realsense color stream extrinsic and intrinsic parameters. Can i get these parameters directly from librealsense or i need calibration the realsense myself?

dorodnic, 2017-07-04:
For intrinsics, you need to configure the device to the required color resolution (using `enable_stream`) and then call `device::get_stream_intrinsics`. The resulting object will provide helper functions to project points to pixels and vice versa, as well as FOV, principal point and focal length parameters. 
For extrinsics, you need to call `get_extrinsics` with the two streams you wish to convert between. For instance, depth and color to get the translation/rotation between depth image and color image.

qwzlh, 2017-07-14:
@Dmytry Thank you very much, that helps a lot!

closed 2017-07-14 by qwzlh