realsenseai/librealsense #543: how can I get aligned depth (depth_aligned_to_color) using asynchronous APIs ?

xxsong5, 2017-06-26:
dev->set_frame_callback(rs::stream::depth_aligned_to_color,  call_back);  //failed!!!  Said that "stream" must be a native stream.

dorodnic, 2017-07-01:
No, using the stable version of librealsense you cannot combine callbacks and synthetic streams.
Callbacks were introduced to provide a solution for low latency usages (such as motion tracking and VR). 
Synthetic streams implicitly require synchronization (to generate single depth_aligned_to_color frame you need to wait for both color and depth frame + processing, introducing considerable latency)

closed 2017-07-01 by dorodnic

ilyak93, 2022-02-23:
@dorodnic, will be any update on this issue ? It sounds as pretty hard constraint on the use of the SDK while the use case seems to be pretty native. 

Is the computation of the alignment that complex and resources consumable ? 
Maybe there is an option to share a version that computes it straight forward without async stuff ?

I'm working on a project where it is much more convenient to use callbacks and that functionality harshly missing.

Thank you in advance.