realsenseai/librealsense #545: After wait_for_frames timeout, get_frame_timestamp gives values between -3 and 4

MFUHH, 2017-06-30:
Sometimes when I start my program, the wait_for_frames function gives a timeout error. I implemented a try...catch handler. In this case I stop the the camera and start it again (dev->stop(); dev->start();). Then the wait_for_frames works, but when I want to give out the timestamps, it gives me values between -3 and 4. The framenumber instead counts regularly.

| Required Info  |   |
|---|---|
| Camera Model |  F200 | 
| Firmware Version |  2.60.00 | 
| Operating System & Version | Windows 10  |

dorodnic, 2017-07-01:
This is rather unusual. If you enable logging and reduce log level (`rs::log_to_console`) do you see anything suspicious in the log? 
F200 camera is passing hardware time of capture using first couple of pixels of the image. (this rather odd method is used in a lack of better alternative before UVC frame metadata) You can usually see these pixels flickering with a naked eye. If they don't, the timestamp will be calculated from whatever pixel values that happened to be in the left-top corner. It is possible this is why you are getting -3 and 4.

MFUHH, 2017-07-05:
Thank you for your advice. I did enable logging and attached the output. In one case the waitForFrames worked from the beginning and the timestamps in this case are right. In the other case, waitForFrames timed out, this was handled with dev->stop(); dev->start(); and the next waitForFrames worked but then the timestamps are wrong. 
I would estimate that in 90% of the cases the waitForFrames works from the beginning. In the other 10 one restart is needed and then it works but with the wrong timestamps. 

[restartNeeded.txt](https://github.com/IntelRealSense/librealsense/files/1124190/restartNeeded.txt)
[workedDireclty.txt](https://github.com/IntelRealSense/librealsense/files/1124191/workedDireclty.txt)

dorodnic, 2017-07-13:
In my opinion, if this problem repeats it is a hardware malfunction. It seems both UVC and the raw USB endpoints get stalled. Please consider reaching out to [RealSense support](http://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html).

closed 2017-07-13 by dorodnic