realsenseai/librealsense #547: Actual exposure time and gain for each fisheye camera frame

NikolausDemmel, 2017-07-04:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |  Camera FW: 2.0.71.28, Adapter FW: 1.29.0.0, Motion Module FW: 1.25.0.0 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux  |

I would like to determine the actual gain and exposure times for the frames of the ZR300's fisheye camera. I am using librealsense `master` branch from this repository.

In order to investigate I modified the `cpp-config-ui` example a little bit to add output for every fisheye frame: exposure time in metadata of the frame, as well as exposure time and gain from `dev->get_options(...)`. I also enabled some output in the fisheye AEC code to see when gain / exposure options are changed.

There are some strange things and unclarities:

1. Whenever exposure time changes, the value embedded in the metadata is about 2x the actual exposure time for one frame. For large exposure times it is sometimes also smaller (maybe some wrap-around).
2. Embedded exposure time is sometimes of by 1 (actually, this is probably the resolution of the metadata embedding, which seems resolved only to 0.2 ms, not 0.1ms like the option).
3. When setting the exposure time, there seems to sometimes be a lag until it takes effect. Which frame does setting the exposure time affect (next one)? Is it the same for gain? Is there any way to know the actual exposure and gain for a given frame?

Here is a section of output with AEC on, followed by some output of changing exposure / gain manually in the gui:

AEC:
```
Reading fisheye frame with exposure metadata 200.0, exposure option 200.0 and gain option 21.0
Reading fisheye frame with exposure metadata 200.0, exposure option 200.0 and gain option 21.0
TotalExposure 56.6, target_exposure 50.5
 ModifyExposure: DecreaseExposureGain: 
 target_exposure0 51.5
 exposure 20.0, gain 2.6
GainModified: gain = 2.6
 rounded to: 2.6
Reading fisheye frame with exposure metadata 400.0, exposure option 200.0 and gain option 19.0
Reading fisheye frame with exposure metadata 200.0, exposure option 200.0 and gain option 19.0
Reading fisheye frame with exposure metadata 200.0, exposure option 200.0 and gain option 19.0
TotalExposure 51.5, target_exposure 44.5
 ModifyExposure: DecreaseExposureGain: 
 target_exposure0 46.8
 exposure 20.0, gain 2.3
GainModified: gain = 2.3
 rounded to: 2.3
Reading fisheye frame with exposure metadata 400.0, exposure option 200.0 and gain option 17.0
Reading fisheye frame with exposure metadata 200.0, exposure option 200.0 and gain option 17.0

[...]

Reading fisheye frame with exposure metadata 110.0, exposure option 110.0 and gain option 15.0
Reading fisheye frame with exposure metadata 110.0, exposure option 110.0 and gain option 15.0
Reading fisheye frame with exposure metadata 110.0, exposure option 110.0 and gain option 15.0
TotalExposure 22.0, target_exposure 23.9
 ModifyExposure: IncreaseExposureGain: 
 target_exposure0 23.9
 exposure 12.0, gain 2.0
output exposure by algo = 12.0
Reading fisheye frame with exposure metadata 110.0, exposure option 119.0 and gain option 15.0
Reading fisheye frame with exposure metadata 220.0, exposure option 119.0 and gain option 15.0
Reading fisheye frame with exposure metadata 118.0, exposure option 119.0 and gain option 15.0
TotalExposure 23.9, target_exposure 26.2
 ModifyExposure: IncreaseExposureGain: 
 target_exposure0 26.2
 exposure 13.1, gain 2.0
output exposure by algo = 13.1
Reading fisheye frame with exposure metadata 118.0, exposure option 130.0 and gain option 15.0
Reading fisheye frame with exposure metadata 238.0, exposure option 130.0 and gain option 15.0
Reading fisheye frame with exposure metadata 130.0, exposure option 130.0 and gain option 15.0
TotalExposure 26.2, target_exposure 28.5
 ModifyExposure: IncreaseExposureGain: 
 target_exposure0 28.5
 exposure 14.3, gain 2.0
output exposure by algo = 14.3
Reading fisheye frame with exposure metadata 130.0, exposure option 130.0 and gain option 15.0
Reading fisheye frame with exposure metadata 260.0, exposure option 142.0 and gain option 15.0
Reading fisheye frame with exposure metadata 142.0, exposure option 142.0 and gain option 15.0
[... 123 lines cut ...]
manually setting fisheye gain to 53.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 53.0
manually setting fisheye gain to 52.0
manually setting fisheye gain to 48.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 48.0
manually setting fisheye gain to 44.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 44.0
manually setting fisheye gain to 36.0
manually setting fisheye gain to 34.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 34.0
manually setting fisheye gain to 29.0
manually setting fisheye gain to 24.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 24.0
manually setting fisheye gain to 22.0
manually setting fisheye gain to 19.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 19.0
manually setting fisheye gain to 15.0
manually setting fisheye gain to 13.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 13.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 13.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 13.0
manually setting fisheye gain to 12.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 12.0
Reading fisheye frame with exposure metadata 300.0, exposure option 300.0 and gain option 12.0
```

dorodnic, 2017-07-04:
Hi @NikolausDemmel!
I'll try to provide the information that I have, and will do my best to get the auto-exposure algorithm author to comment. 

So, regarding item 2:
It is surprisingly tricky to provide custom values with each frame of a UVC device. With UVC 1.5 spec, the closest thing to "standard" is per-frame metadata. This feature was introduced in Windows 10 (you can get some info [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/uvc-extensions-1-5) and for Linux, Intel is actively involved in getting this functionality [up-streamed](https://patchwork.linuxtv.org/patch/38581/)) That said, the alternatives are to embed data into pixels (F200) or expand the frame by a couple of lines (R200). 
ZR300 is embedding 4 bit frame number into the least significant bits of the first pixels to minimize visual artifacts and 8 bit (if I recall correctly) actual exposure. The resulting exposure has one bit less then the corresponding control, due to processing limitations in firmware. 

Regarding item 3:
I believe exposure and gain take affect after two frames. In addition, you should take into account USB and V4L subsystems latency. Usual latency of fish-eye stream is about 15 milliseconds, but if the processing is slow and the host is busy, we allocate up to 4 extra frames for V4L2 to use. This can cause extra latency. 

There are most likely ways to optimize the convergence rate (we experimented with a few) but overall the algorithm was designed with a specific usage in mind: motion tracking with SLAM. For this task, gradual convergence is often better because it makes it easier to track features in the image. May I ask what is the use case you have in mind?

NikolausDemmel, 2017-07-04:
Hi @dorodnic, thanks for the quick reply!

> So, regarding item 2:
It is surprisingly tricky to provide custom values with each frame of a UVC device. With UVC 1.5 spec, the closest thing to "standard" is per-frame metadata. This feature was introduced in Windows 10 (you can get some info here and for Linux, Intel is actively involved in getting this functionality up-streamed) That said, the alternatives are to embed data into pixels (F200) or expand the frame by a couple of lines (R200).
ZR300 is embedding 4 bit frame number into the least significant bits of the first pixels to minimize visual artifacts and 8 bit (if I recall correctly) actual exposure. The resulting exposure has one bit less then the corresponding control, due to processing limitations in firmware.

Ok, got it. So I guess my item 1. is a bug in the firmware or driver?

> Regarding item 3:
I believe exposure and gain take affect after two frames. In addition, you should take into account USB and V4L subsystems latency. Usual latency of fish-eye stream is about 15 milliseconds, but if the processing is slow and the host is busy, we allocate up to 4 extra frames for V4L2 to use. This can cause extra latency.

Ok, understood. So there is currently no way to do this reliably, but approximately by checking the options and possibly wating a frame or two. It would be super helpful if the gain as a per-frame value could maybe be included in future versions of the firmware / driver.

> There are most likely ways to optimize the convergence rate (we experimented with a few) but overall the algorithm was designed with a specific usage in mind: motion tracking with SLAM. For this [...]

You are talking about convergence of the AEC, right? Note that what I am observing is somewhat independent of AEC and also happens with AEC off. But in any case, my use case is precisely SLAM. However, I would like to use photometric techniques (https://github.com/JakobEngel/dso), which don't use feature points. DSO for instance tries to compensate for changes in exposure and gain to estimate the true illuminance of the scene and compare that between frames. This way sudden brightness changes are also not a problem, but it relies on knowing the exposure times and gain. However, when using AEC, as you say the exposure changes only slowly, so being off by a frame might be suboptimal, but tolerable.

NikolausDemmel, 2017-07-12:
@dorodnic, do you happen to have any update on the potential bug in the firmware (item 1.)?

When you say about per-frame metadata in the UVC1.5 spec that "Intel is actively involved in getting this functionality up-streamed", does that mean that you plan to make use of that in the Realsense driver in the future? Are there any concrete plans to provide per-frame exposure-related info such as shutter and gain, or is this very much hypothetical?

dorodnic, 2017-07-13:
SR300 (Close-Range) and upcoming RS400 (Long-Range) both provide actual exposure and gain through UVC metadata. Once this functionality is upstreamed we will align to the standard. Until then we will make proprietary pacthes available for 4.4, 4.8 and 4.10. 
It seems that ZR300 firmware is pretty much "sealed" at this time so it is hard to get a comment. You can however submit official support request through the [support site](http://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html).

NikolausDemmel, 2017-07-13:
Thanks for the info, but the SR300 and RS400 both don't feature a fisheye tracking camera, right?

About the possible bug with the metadata I contacted the support as suggested.

jlgarcia75, 2017-09-12:
@NikolausDemmel, we are keeping this issue open here on GitHub until we receive a response. I will mark this thread for follow up.

Jesus
Intel Customer Support

NikolausDemmel, 2017-09-12:
After some back and forth with customer support, they could reproduce the issue and sent a report to engineering. I'm still waiting on their response but will try to update this issue when I have news.

jlgarcia75, 2017-10-03:
Hello @NikolausDemmel,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.  

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75

NikolausDemmel, 2017-10-04:
@jlgarcia75,

thanks for the update, it is much appreciated.

one question regarding the replacement with the D400 series. So far I only found info on the D415 and D435, neither of which seems to have IMU and fisheye tracking camera.

Do you know if we can expect a variant in the D400 series that is similar to the ZR300 in that respect?

Thanks,
Nikolaus

jlgarcia75, 2017-10-04:
@NikolausDemmel,

The current D400 series cameras will not have the IMU or fisheye. Unfortunately, I cannot comment on whether one will come out in the future that has those features.

Regards,
Jesus
Intel Customer Support

Shivani1796, 2019-09-05:
@NikolausDemmel  Hello, can you help me to find exposure time of my fisheye image, when I check properties of the image I can find written exposure time but not the value so how to figure out exposure time of each image.
Thanks in advance

NikolausDemmel, 2019-09-05:
No, sorry, haven't used this camera in a long time. I don't know what the state of the SDK is these days.

Shivani1796, 2019-09-06:
Hey, do you know about pcalib.txt files which is used in DSO dataset. (
https://github.com/JakobEngel/dso ) I am working to make my own dataset
from that i need that response callibration file  & pcalib.txt file do you
know about that ?
Regards
Shivani Baldwa
Ming Chi University Of Technology
- 84 Gungjuan Rd, Taishan Dist, New Taipei City 24301
-
-
-
-
-
-
-
-  Taiwan

NikolausDemmel, 2019-09-06:
Its generated by the `responseCalib` tool in that respository: https://github.com/tum-vision/mono_dataset_code

Please refer to the code if you want to generate it differently. Or else also the DSO code to see how it is used there.

I cannot help you more, and this is really off-topic here.

Shivani1796, 2019-09-06:
Thanks for your reply, won't bother you from now.
Thanks
Regards
Shivani Baldwa
Ming Chi University Of Technology
- 84 Gungjuan Rd, Taishan Dist, New Taipei City 24301
-
-
-
-
-
-
-
-  Taiwan