realsenseai/librealsense #548: Units of gain and exposure for Fisheye camera

NikolausDemmel, 2017-07-04:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |  Camera FW: 2.0.71.28, Adapter FW: 1.29.0.0, Motion Module FW: 1.25.0.0 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux  |

From https://github.com/IntelRealSense/librealsense/blob/2a062ab6d31a0f8917908492533a9dfc9bba0dcb/src/zr300.cpp#L677-L693 I am deducing that the units for exposure and gain options are:

* exposure: `exposure_seconds = exposure_option / 10000`
* gain: `gain_factor = (exposure_option - 15) / 8 + 2`

Can someone confirm this?

Is `gain_factor` as above a simple multiplicative factor on the intensity values? I.e. if `gain_factor` was `2.0`, the pixel value is twice as bright as it would be without any amplification? I assume the gain factor in the imager is applied before gamma encoding of the value, i.e. in a static scene, a `gain_factor` of 2.0 has the same effect on pixel intensities as doubling the exposure time (except that it also amplifies the noise).

See also my recent question on retrieving the gain and exposure time values: #547

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello @NikolausDemmel,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75

NikolausDemmel, 2017-10-17:
@jlgarcia75, sorry I didn't respond directly.

I guess the above questions are still unconfirmed. I haven't found a reason to believe my assumptions are wrong though...

jlgarcia75, 2017-10-17:
Hi @NikolausDemmel,

I'm sorry to say that the ZR300 camera is no longer supported so there is no way for me to confirm your assumptions or answer your questions. Therefore, I will leave this issue closed.

Regards,
Jesus G.
Intel Customer Support

oordan, 2017-10-18:
Hi @NikolausDemmel,
You are correct about the exposure units. 
Gain units are different. It is supposed to be `gain_factor = gain_value / 16` for the imager.
The code you refer to above is SW auto exposure implementation.

NikolausDemmel, 2017-10-18:
@oordan, thanks for following up!

About the gain units: Are you sure? Your suggestion does not add up with the SW auto exposure /  auto gain control implementation of the fisheye camera that I linked above, or am I missing something?

```
 double value[] = { (gain_value-2) * 8 +15. }; 
```

and somewhere else

```
float total_exposure = exposure * gain
```

where `gain` is set to `gain_value` above, meaning that the `gain_value` in the code, is what you refer to as `gain_factor` in your comment.

oordan, 2017-10-18:
Unfortunately this is the best assumption I could get from HW dev here, but no confirmation.
The SW doesn't modify the gain value written to the gain register, and the common sensors behavior is as I supposed above. I couldn't get info if this sensor behaves differently, or if the code above might be wrong.

NikolausDemmel, 2017-10-18:
Alright. Thanks for investigating. It is much appreciated even if not 100% clarified.