realsenseai/librealsense #549: error while running the patching script

AnirudhPuligandla, 2017-07-05:
| Required Info  | I was following the installation instructions for librealsense provided at [docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) . But while running the script to download, patch and build uvc kernel module from sources, i get an error as follows:

[image]
 |
|But the modprobe command executes without an error.|Should i consider the module to be installed successfully or is the error encountered at the patching step could be a potential issue?|
| Camera Model | R200 | 
| Firmware Version |  v1.12.1 | 
| Operating System & Version | ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-81-generic  |

ev-mp, 2017-07-05:
Hi, you didn't specify the exact configuration, but according to the screenshot there seem to be discrepancy between the linux headers installed (Line7: 4.4..0-83) and the kernel running (last line: 4.4.0-81).

Please check the exact kernel version ("uname -r") and if it is not 4.4.0-83 then update distribution according to [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#make-ubuntu-up-to-date)  :
"$sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade".
Make sure to reboot and select the appropriate kernel version before running the patch again.

AnirudhPuligandla, 2017-07-06:
Thank you @ev-mp for the suggestion. The patching process now worked without any issues.

closed 2017-07-06 by AnirudhPuligandla