realsenseai/librealsense #551: Write PPM/PGM files

mbuffier, 2017-07-14:
| Required Info  |   |
|---|---|
| Camera Model |  ZR300 | 
| Firmware Version |   | 
| Operating System & Version | Linux  |
| Kernel Version (Linux Only)  | Ubuntu 16.04  |

Hi, 

I'm trying to obtain ppm and pgm files ( ZR300 camera) from the depth and color  stream to be able to use them after. Any idea how should I do that ? (or any file format with which I can store the depth value directly without scaling). 

Thank you, 

Maud

mojovski, 2017-07-14:
you really should not do it. :)
lets assume you use float values for each pixel in the image to represent the depth directly: this makes 32 bit per each pixel. Compared to 8 bit, this is 4 times more. 
So lets assume you hav 640x480 8bit images, which makes 2492160 bytes (2.433 MB) per each image.
Using float would result in 9.7 MB per image.
And if you make a video with only 3 fps, this would give you 1.7GB instead of  427 MB .

Ok, you might say, I dont care. I have enough storage...
but reading such large data files also requires time. On a standard HDD you would need 500ms to read one single image with 32 bit and only 120ms for a 8 bit one. 

However, I fully understand that in some cases it is necessary to use more bits for depth. Especially when you deal with remote sensing or high precision stuff...

What you might do is the following:

* use 16 bit.
* scale it in a non-linear fashion. Perspective cameras have higher resolutions for 3d distances closer to the camera and coarser resolution further away. So you might expose it. Consider this: Lets assume you have a depth pixel value of 5. What distance difference would result to change the pixel value to 6?
Lets just say 1mm.
So, consider a depth value difference between 54000 and 54001 which represents an object far away from the camera. Lets say far away would mean 4m (for a standard rgbd camera its a fair max distance).
The resolution of the sensor is at this distance somewhat about 10cm. So it would would be a waste of storage to scale the depth difference between 54000 and 54001 to 1mm. Scale it to 10cm then. :)

If you like this idea and need some help implementing this, let me know.

I hope this helps.

PS: Regarding the data format output from the ZR300 I am not sure what is supported for now. I also git one some days ago. But I saw that they also deliver pure 3d points as a ros topic, which you might want to project on an image plane and save as a depth image.

jlgarcia75, 2017-09-12:
Hi @mbuffier , do you still need further help on this issue?

Jesus
Intel Customer Support

freemanlo, 2017-09-21:
Hi @**muffier** , you can try to use OpenCV imwrite function to save the depth image into the PGM or 16bit PNG file. Then, you can use imread and imshow to display the image. The demo code as follows:

	cv::Mat depth16(_depth_intrin.height, _depth_intrin.width, CV_16U, (uchar *)_rs_camera->get_frame_data(rs::stream::depth));

	imwrite("depth16.png", depth16);
	imwrite("depth16.pgm", depth16);

	Mat depthMap1 = imread("depth16.png", CV_LOAD_IMAGE_ANYDEPTH);
	imshow("depthMap1", depthMap1);
	Mat depthMap2 = imread("depth16.pgm", CV_LOAD_IMAGE_ANYDEPTH);
	imshow("depthMap2", depthMap1);

jlgarcia75, 2017-09-29:
Thank you, Freeman. Closing this case. Status: Abandoned by Customer.

Jesus
Intel Customer Support

closed 2017-09-29 by jlgarcia75