realsenseai/librealsense #553: Why are depth and color image out of time sync?

mojovski, 2017-07-14:
| Required Info  |   |
|---|---|
| Camera Model |  ZR300 | 
| Firmware Version |   | 
| Operating System & Version |   Ubuntu 16.04 LTS|
| Kernel Version (Linux Only)  |  4.4 |

Hi, I just noticed that the data coming from the ZR300 is out of sync. 
I understand that the IMU readings are of different times (which is a pitty) but I dont get the reason why the color and the depth images have different time stamps.
Below is a screenshot from the rqt_bag showing the color and depth topics on a time line.

[image]

Can someone give me a hint? 
Or: maybe I did something wrong in the acquisition process and this is a mistake on my side?

Thanks alot!

dorodnic, 2017-07-18:
How do you measure sync? Do you mean that the frames have different reported timestamps or the timestamps are the same and visually the images don't match? 
Another factor is color resolution. To the best of my knowledge the firmware is treating HD color differently. What is your configuration?

NikolausDemmel, 2017-08-03:
>  Below is a screenshot from the rqt_bag showing the color and depth topics on a time line.

I don't think you are looking at the timestamps you want to be looking at. The timestamps on the timeline in rqt_bag are the rosbag timestamps, i.e. the time when the messages were recorded. Those are not the same as the image timestamps reported by the driver that can be found in the message header.

Naturally, the rosbag timestamps will almost never be identical, even for a perfectly synchronized sensor.

Now, this doesn't mean that the images ARE synchronized, but you should look at the `header.stamp` timestamps in the messages rather than the bag timestamps.

mojovski, 2017-08-08:
@NikolausDemmel , thanks! Thats exactly the issue. 
One need to look at the header stamps and not on the data shown in the gui...

closed 2017-08-08 by mojovski