realsenseai/librealsense #554: Issue when running ./scripts/patch-realsense-ubuntu-xenial.sh

vijayemmanuel90, 2017-07-15:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model | R200
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16. 04.2 LTS |
| Kernel Version (Linux Only)  |  4.8.0-58-generic |

I ran into the below issue when trying to build uvc kernel modules
```
./scripts/patch-realsense-ubuntu-xenial.sh

patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 188 with fuzz 2 (offset 20 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 143 (offset 12 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 624.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
```

dorodnic, 2017-07-18:
There has been recent changes to Ubuntu default kernel, we will publish an update to the patch shortly.

ev-mp, 2017-07-19:
Done. 
The kernel source ppa for version 4.8 was remapped based on [PR#546](https://github.com/IntelRealSense/librealsense/pull/546)

closed 2017-07-19 by ev-mp