realsenseai/librealsense #555: /usr/local/lib/librealsense.so: undefined reference to `libusb_strerror'

raymond00000, 2017-07-19:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model |  SR300  | 
| Firmware Version |   Firmware version: 3.21.0.0 | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.4.0-83-generic |

I built the latest source from github. I can run the example-tutorial. 

But I cannot make the bin by following this tutorial:
https://software.intel.com/en-us/articles/using-librealsense-and-pcl-to-create-point-cloud-data

[ 50%] Linking CXX executable LRS_PCL
/usr/local/lib/librealsense.so: undefined reference to `libusb_strerror'
/usr/local/lib/librealsense.so: undefined reference to `libusb_get_port_numbers'
/usr/local/lib/librealsense.so: undefined reference to `libusb_get_parent'
collect2: error: ld returned 1 exit status
CMakeFiles/LRS_PCL.dir/build.make:290: recipe for target 'LRS_PCL' failed
make[2]: *** [LRS_PCL] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/LRS_PCL.dir/all' failed
make[1]: *** [CMakeFiles/LRS_PCL.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

raymond00000, 2017-07-19:
solved, I installed another sensor with libusb in its lib, but that libusb is old version, and the cmake used that old version. Thanks.

closed 2017-07-19 by raymond00000

wf-hahaha, 2020-03-28:
@raymond00000 hello，Did you solve it? how to solve it？