realsenseai/librealsense #556: Could NOT find RSSDK (missing:  RSSDK_LIBRARIES RSSDK_INCLUDE_DIRS)

raymond00000, 2017-07-19:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |   |

I downloaded PCL1.8, and ran cmake ..
I got below message. what should i do to setup the path of RSSDK. Thanks.

-- Could NOT find RSSDK (missing:  RSSDK_LIBRARIES RSSDK_INCLUDE_DIRS) 
CMake Warning at cmake/pcl_targets.cmake:864 (message):
  RealSense SDK support: not building because RSSDK not found
Call Stack (most recent call first):
  CMakeLists.txt:302 (PCL_ADD_GRABBER_DEPENDENCY)

jlgarcia75, 2017-09-12:
Hi, please be more specific on what you downloaded and steps you took when this issue occurred. Give us more details so we can reproduce your issue.

Jesus
Intel Customer Support

jlgarcia75, 2017-09-29:
Closing case. Status: Customer abandoned.

closed 2017-09-29 by jlgarcia75