realsenseai/librealsense #557: Extracting single frame 3D models from the SR300

praveenlovely, 2017-07-19:
Hi team,

Do you have samples to extract single frames from the camera?

| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version |  3.21.0.0 | 
| Operating System & Version | Windows 10   |

dorodnic, 2017-07-24:
Well, it depends on what you mean by single 3D frames. RealSense cameras do not provide "stills", you must configure the device for video stream. [cpp-headless](https://github.com/IntelRealSense/librealsense/blob/master/examples/cpp-headless.cpp) is saving single frame on each stream-type as a PNG image. You can also modify the [pointcloud](https://github.com/IntelRealSense/librealsense/blob/master/examples/cpp-pointcloud.cpp) demo to save the 3D model into PLY / OBJ format.

praveenlovely, 2017-07-25:
@dorodnic Thanks for your comment.
This clarifies some of the issues. Do you have a sample code that outputs the 3D Model as PLY/OBJ?

dorodnic, 2017-07-28:
At the moment we don't, but it is in the backlog

praveenlovely, 2017-07-28:
Thanks for the information.
It would be helpful if you can provide any documentation/guide/examples for
any facial scanning.

Thanks and regards,
Praveen Kumar P.S

oordan, 2017-08-10:
Unfortunately we currently don't have facial scanning examples.

closed 2017-08-10 by oordan