realsenseai/librealsense #558: R200 VIDIOC_DQBUF error 19, No such device

raymond00000, 2017-07-20:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.10 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-83-generic  |

I can install and ran the example code with SR300, but I got error for R200. 
I applied this script. 
https://github.com/IntelRealSense/librealsense/blob/master/scripts/install-r200-udev-fix.sh

a@a-desktop:/usr/local/bin$ ./cpp-tutorial-3-pointcloud 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense R200
    Serial number: 
    Firmware version: 1.0.72.10
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)

ev-mp, 2017-08-10:
Hi @raymond00000 ,
The specific error type occurs when host-device connection is interrupted.
This often occurs due to insufficient power or other hw-related issues.

There is a substantial volume of technical info gathered and evaluated by user in previous threads.
You may find useful info looking into threads, such as [515](https://github.com/IntelRealSense/librealsense/issues/515), [437](https://github.com/IntelRealSense/librealsense/issues/437), [418](https://github.com/IntelRealSense/librealsense/issues/418) and 
[108](https://github.com/IntelRealSense/librealsense/issues/108).

closed 2017-08-21 by ev-mp

mgcrea, 2017-09-04:
I'm encountering the exact same issue and I'm using an Anker USB powered hub so it's not power related:

```
[  484.815148] WARNING: CPU: 0 PID: 11439 at /build/linux-hwe-edge-sLLR41/linux-hwe-edge-4.11.0/fs/sysfs/group.c:237 sysfs_remove_group+0x8c/0x90
[  484.815150] sysfs group 'power' not found for kobject 'media1'
[  484.815151] Modules linked in: iTCO_wdt iTCO_vendor_support rfcomm ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo iptable_nat nf_nat_ipv4 br_netfilter bridge stp llc aufs xt_multiport bnep hid_generic snd_hda_codec_hdmi nls_iso8859_1 snd_soc_skl snd_soc_skl_ipc snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_sst_match snd_soc_core snd_hda_codec_realtek snd_hda_codec_generic snd_compress ac97_bus snd_pcm_dmaengine snd_usb_audio usbhid snd_usbmidi_lib snd_hda_intel snd_hda_codec arc4 snd_hda_core snd_hwdep snd_pcm intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_seq_midi snd_seq_midi_event snd_rawmidi kvm irqbypass crct10dif_pclmul snd_seq crc32_pclmul ghash_clmulni_intel pcbc iwlmvm snd_seq_device snd_timer mac80211 aesni_intel
[  484.815209]  aes_x86_64 crypto_simd snd glue_helper cryptd iwlwifi intel_cstate intel_rapl_perf hci_uart btusb btrtl rtsx_pci_ms ir_rc6_decoder btbcm cfg80211 memstick soundcore btqca rc_rc6_mce btintel ir_lirc_codec uvcvideo lirc_dev ite_cir bluetooth videobuf2_vmalloc rc_core videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media acpi_als intel_lpss_acpi intel_lpss kfifo_buf mei_me industrialio shpchp tpm_crb mei mac_hid intel_pch_thermal acpi_pad ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 xt_hl ip6t_rt nf_conntrack_ipv6 nf_defrag_ipv6 ipt_REJECT nf_reject_ipv4 nf_log_ipv4 nf_log_common xt_LOG xt_limit xt_tcpudp xt_addrtype nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack ip6table_filter ip6_tables nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack i2c_smbus
[  484.815278]  libcrc32c i2c_i801 iptable_filter parport_pc ip_tables ppdev x_tables sunrpc lp parport autofs4 i915 rtsx_pci_sdmmc i2c_algo_bit drm_kms_helper syscopyarea e1000e sysfillrect sysimgblt fb_sys_fops nvme ptp nvme_core pps_core drm rtsx_pci ahci libahci wmi video pinctrl_sunrisepoint pinctrl_intel i2c_hid hid
[  484.815313] CPU: 0 PID: 11439 Comm: c-tutorial-2-st Tainted: G        W       4.11.0-13-generic #19~16.04.1-Ubuntu
[  484.815315] Hardware name:                  /NUC7i3BNB, BIOS BNKBL357.86A.0045.2017.0417.1728 04/17/2017
[  484.815316] Call Trace:
[  484.815320]  dump_stack+0x63/0x90
[  484.815323]  __warn+0xcb/0xf0
[  484.815327]  warn_slowpath_fmt+0x5a/0x80
[  484.815331]  sysfs_remove_group+0x8c/0x90
[  484.815336]  dpm_sysfs_remove+0x57/0x60
[  484.815340]  device_del+0x12d/0x350
[  484.815348]  media_devnode_unregister+0x3f/0x60 [media]
[  484.815353]  media_device_unregister+0x147/0x180 [media]
[  484.815359]  uvc_delete+0x6d/0x160 [uvcvideo]
[  484.815365]  uvc_release+0x23/0x30 [uvcvideo]
[  484.815375]  v4l2_device_release+0xe6/0x120 [videodev]
[  484.815380]  device_release+0x32/0x90
[  484.815384]  kobject_release+0x6a/0x170
[  484.815388]  kobject_put+0x2f/0x60
[  484.815392]  put_device+0x17/0x20
[  484.815401]  v4l2_release+0x50/0x80 [videodev]
[  484.815406]  __fput+0xe7/0x220
[  484.815411]  ____fput+0xe/0x10
[  484.815416]  task_work_run+0x7e/0xa0
[  484.815420]  do_exit+0x2e2/0xb60
[  484.815424]  ? hrtimer_cancel+0x19/0x20
[  484.815427]  do_group_exit+0x43/0xb0
[  484.815431]  get_signal+0x293/0x620
[  484.815436]  do_signal+0x37/0x750
[  484.815439]  ? do_futex+0x210/0x530
[  484.815443]  ? __check_object_size+0x108/0x19e
[  484.815449]  exit_to_usermode_loop+0x76/0xb0
[  484.815453]  syscall_return_slowpath+0x59/0x60
[  484.815457]  entry_SYSCALL_64_fastpath+0xab/0xad
[  484.815459] RIP: 0033:0x7fd92737c709
[  484.815461] RSP: 002b:00007ffe2362b850 EFLAGS: 00000202 ORIG_RAX: 00000000000000ca
[  484.815464] RAX: fffffffffffffdfc RBX: 0000000000976550 RCX: 00007fd92737c709
[  484.815466] RDX: 0000000000000017 RSI: 0000000000000189 RDI: 00000000009883bc
[  484.815468] RBP: 000000000095d120 R08: 000000000095d130 R09: 00000000ffffffff
[  484.815470] R10: 00007ffe2362b910 R11: 0000000000000202 R12: 0000000000000017
[  484.815472] R13: 00007ffe2362b910 R14: ffffffffffffff92 R15: 00007ffe2362b800
[  484.815476] ---[ end trace bcc18219e8acab32 ]---
```

@ev-mp, Should I open a new issue?

ev-mp, 2017-09-05:
@mgcrea , the power consumption is definitely one of the first things to look into, but not only:
the quality and length of USB3 cables matters as well. The depth module itself may be faulty.

In case the disconnects are deterministic and can be reproduced by application/code - you're welcomed to open new issue and we will definitely address it.

Otherwise you need to check the HW setup (R200+USB3 cable + USB3 hub) and probably replace some components to isolate the weakest link.
Additional hints/things to consider:
 - Are you using the original (30cm) USB3 cable supplied with R200?
 - Does this specific HW setup works/fails when connected some other PC/OS ?
 - Did you try to use Ubuntu stable (LTS) kernel instead of 4.11.0-13?