realsenseai/librealsense #559: Is camera/depth/image_raw already alilgned to RGB?

mojovski, 2017-07-21:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | ?  | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.4 |

I just would like to know if the depth data coming in the ros topic camera/depth/image_raw
is already aligned to RGB or not. 

For now I am experiencing large displacements but this could also be the result of bad extrinsics provided by the hardware off-the-shelf.

Thanks in advance!!

freemanlo, 2017-09-12:
If ZR300 depth is calibrated, the alignment based on RGB-to-depth extrinsics should be there. Did you try to do UVMap between depth and RGB and find the alignment issue on your ZR300 device?

jlgarcia75, 2017-09-12:
I see that @mojovski got help on this issue in the [RealSense Community](https://communities.intel.com/thread/116518) a few months ago. Hence I will close this issue on here.

Jesus
Intel Customer Support

closed 2017-09-12 by jlgarcia75