realsenseai/librealsense #562: uvcvideo failes to insert patchd module

PaoloCorazza, 2017-07-27:
Hello guys,

I am installing librealsense from scratch and I am facing this problem with the uvc module. What's wrong?
My kernel is `4.10.0-27-generic`

```
make: Leaving directory '/home/asd/librealsense-master/ubuntu-xenial-hwe'
Patched kernel module created successfully

Handle uvcvideo :
Applying the patched module ... modprobe: ERROR: could not insert 'uvcvideo': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  uvcvideo  module was reloaded
```

| Required Info  | UVC module |
| Camera Model | SR300 | 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  | 4.10.0-27-generic  |

dorodnic, 2017-07-28:
Hi @PaoloCorazza 
At the moment we provide patches for 4.4 and 4.8 kernels. 
We have 4.10 patch in the working, but it will not be published for at least a week or two. 
So, for now your options are to downgrade to one of the supported kernels, upgrade to 4.12 RC (you don't need patches there) or modify the patch yourself. 
We will update the issue once we have a solution

PaoloCorazza, 2017-07-31:
[image]

Thanks, @dorodnic. I solved the problem choosing a different kernel thanks to "Grub customizer", which helped me to solve the problem in a quite straightforward way.

ev-mp, 2017-08-10:
Hi @PaoloCorazza ,
The patch for Ubuntu 16.04.2 [was updated](https://github.com/IntelRealSense/librealsense/pull/569) as Canonical completed transition from kernel 4.8 to 4.10.
You need to updated your Ubuntu to the latest *4.10.0-30+), then rerun the patch

closed 2017-08-10 by dorodnic

LZhipeng, 2017-12-15:
Hi, @ev-mp ,
How to patch on ubuntu 16.04.3 with Linux kerel 4.10.0-42-generic?

Thanks,
Zhipeng