realsenseai/librealsense #563: Why should uvcvideo be a module (rather than a built-in)?

Maghoumi, 2017-07-30:
My question is not specific to a sensor, but rather the library itself (thus I'm not including the device/model tags at the top of my question).

I noticed that the latest SDK release for NVIDIA Jetson TX1/TX2 boards (which was released only a few days ago) uses kernel v4.4 and the sources already contain the necessary patches for RealSense video formats, etc. 

Right out of the box, I noticed that upon connecting my R200 to my Jetson TX1 board, `dmesg` indicated that the connected camera was correctly recognized: 

> Found UVC 1.10 device Intel RealSense 3D Camera R200

I could also confirm that `uvcvideo` was already available as a builtin. I proceeded with building the latest librealsense source. However, I noticed that upon trying to run any of the samples I would get:

> uvcvideo kernel module is not loaded

I went ahead and built the kernel from the sources, from scratch. Except for forcing `uvcvideo` to be built as a module, I did not touch any of the other settings. Now the camera and the samples work fine without any issues.

Forgive my ignorance (in case this is a stupid question) but is there a reason that `uvcvideo` should be built as a module for librealsense to work?

dorodnic, 2017-07-31:
Hi @Maghoumi 
To the best of my knowledge there is no such limitation. When the pixel-formats are included in the Kernel, RealSense cameras function just like any other V4L2 streaming device.
When you were using the built-in drivers, did you notice anything unusual in `dmesg` when connecting the device? Was the device visible in `lsusb | grep 8086` output? Was it connected to the USB3 port in the same way (using same hub and cables)?

dorodnic, 2017-07-31:
I just recalled we have the following check in the V4L2 backend:
[uvc-v4l2.cpp#L728-L744](https://github.com/IntelRealSense/librealsense/blob/master/src/uvc-v4l2.cpp#L728-L744)
This is a legacy precaution that does not contribute to normal operation of the library. We have tested internal releases of the library without it, so it can be safely removed.

Maghoumi, 2017-07-31:
@dorodnic Thanks. 

I removed those lines and I can confirm that R200 works out of the box on a Jetson TX1 that's is newly flashed with Jetpack v3.1. For some reason, I didn't think of going into the code to see where that error message "uvcvideo kernel module is not loaded" was being thrown.

I'm guessing you'll be replacing those checks with something that can also check for builtin modules?

dorodnic, 2017-08-02:
Actually, internal RS400-version of librealsense works just fine without such checks at all. The team has added a lot of extra checks into the public patching scripts / DKMS, with the ultimate goal of working on upstream kernels. A more practical way to double check that the kernel indeed supports RealSense is querying supported formats using `ioctl(_fd, VIDIOC_ENUM_FMT, &pixel_format)` and examining the resulting `pixel_format`. 
If you would like, you are welcomed to submit a change (see [CONTRIBUTING.md](https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md)), it can save time for future TX1 users. 
Take into account that librealsense is not optimized for Jetson. I just got one recently. There might be existing issues. Please share your experience.

Maghoumi, 2017-08-02:
@dorodnic Thanks for letting me know. Sounds good, I will look into that when I have time and (hopefully) submit a pull request with the necessary changes.

I've been a long time RealSense + TX1 user [(as discussed here)](https://github.com/IntelRealSense/librealsense/issues/73). I've never faced any issues with my R200 on TX1. The performance is mostly good. 

As a matter of fact, for a personal project, I made small changes to probe for CUDA and if available, use CUDA forYUY2 to RGB conversion. The reason I did that was because I noticed the CPU-based conversion in the absence of SSSE3 was causing a bottleneck on TX1 with my particular application. That implementation is available [in a fork I created a while back](https://github.com/Maghoumi/culibrealsense). I didn't know if you guys had any interest to support CUDA if available on the target machine, so I never bothered creating a proper pull request.

If there's any interest in that, I may work on adding proper support for CUDA as well.

ev-mp, 2017-08-10:
We've [completely removed the section](https://github.com/IntelRealSense/librealsense/pull/570) since checking the presence of uvcmodule was found ineffective, and also introduced additional constrain on the kernel configuration.
We've also experimented with an [alternative approach](https://github.com/IntelRealSense/librealsense/pull/442), but eventually it was abandoned as well in favor of removing the test altogether.

closed 2017-08-10 by ev-mp