realsenseai/librealsense #565: SR300 depth distortion

kyranf, 2017-08-04:
Camera Model = SR300 dev kit
 Firmware Version = 3.21.0.0
Operating System & Version = Windows 10 Home, Version 1703 (build 15063.483)

Getting the depth frame at 640x480x30fps from the SR300 and then mapping it into an OpenCV matrix of Uin16_t type,  saving as a tif and showing it in ImageJ 3D visualiser makes it very clear there is pretty bad distortions in the depth image. The image shown is of an object on a flat table surface (white) with a slight tilt (I couldn't hold it perfectly parallel with the table surface..). 

The curly bits on the corners should not be there, and represent in worst cases 25mm of incorrect depth. 

How can I fix this? Additional calibration? Librealsense API calls i'm not understanding or not aware of?  Thanks for any help getting better quality from the SR300. 

[image]

jlgarcia75, 2017-09-12:
The effective range for the SR300 camera is reported as 20 cm to 1.5 meters. According to your environment description, the camera is placed 20 cm from the table and the object is 42mm tall. You may be placing the camera too closely to get good depth readings. Try placing the camera a few more centimers away. If this doesn't help, you may have an uncalibrated camera and need to return it to the place of purchase.

kyranf, 2017-09-12:
During some field-testing, I got reasonable data having the sensor placed 30cm off the flat surface, and the ~5cm tall objects being scanned seemed to be well-formed. There was still up to 10mm worth of curling at the edges of the image with these datasets, where the points should have followed a flat plane.

jlgarcia75, 2017-09-12:
We have seen these issues before when the SR300 camera is not calibrated correctly. If changing the distance does not help sufficiently then you may need a new camera. If you purchased your camera at http://click.intel.com then you have to email click.support@intel.com to get a replacement.

Jesus
Intel Customer Support

kyranf, 2017-09-12:
Okay, i'll mention this to my boss. The Euclid I have appears to give great data, no obvious curling data at the edges, so I guess mileage may vary hehe.  Thanks Jesus!

jlgarcia75, 2017-09-29:
Closing case.

closed 2017-09-29 by jlgarcia75