realsenseai/librealsense #566: Lunar release

mikaelarguedas, 2017-08-05:
Howdy @dorodnic @mdhorn and librealsense maintainers!

A few months ago the new version of ROS, Lunar Loggerhead, [has been released](https://discourse.ros.org/t/ros-lunar-loggerhead-release/1870).
More and more ros projects are using realsense so releasing the drivers on Lunar will unblock users [currently not able](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=librealsense&s=6&r=1) to release their packages.

Thanks!

omershalev, 2017-08-07:
As a temporary solution you can build the package from sources. I did the following:
1. Followed the instructions in http://wiki.ros.org/realsense_camera/Tutorials/Building_from_Sources
2. Cloned librealsense from https://github.com/IntelRealSense/librealsense/ under the same catkin workspace
3. catkin_make

For getting the pointcloud I also had to install ros-lunar-rgbd-launch from debian.

mikaelarguedas, 2017-08-07:
thanks @omershalev for the pointers!
My goal here is to get more widely used packages released into lunar so that users can easily use and release their packages on top.

prarobo, 2017-09-05:
Is there going to be a ros realease for the latest release of librealsense?

mikaelarguedas, 2017-09-07:
@icarpis @dorodnic @mdhorn friendly ping

dorodnic, 2017-09-07:
D400 software release is going to include updated ROS wrapper (very soon). 
We are also updating and validating the R200 wrapper with pending fixes, and planning to handle Lunar. However, D400 is currently the priority.

mikaelarguedas, 2017-09-12:
@jlgarcia75 Has this been closed because the package has been released ? :stuck_out_tongue:

jlgarcia75, 2017-09-12:
Hello @mikaelarguedas, no the package has not been released as @dorodnic specified. Do you have any further questions we can help you with?

mikaelarguedas, 2017-09-12:
Thanks @jlgarcia75  for your quick answer 

> Do you have any further questions we can help you with?

No further questions I just thought that the issue was closed because it has been resolved.
Feel free to close this if it follows your procedure. If you do so, is there is another thread I should follow to get notified when the package gets released?

closed 2017-10-03 by jlgarcia75

mikaelarguedas, 2018-06-05:
@dorodnic very gentle :bellhop_bell: 
Is there any update on this ?
I see that v2 has been out for some time, is there a pan to release v2 in ROS? (If yes, do you think it could be released in ROS Lunar? or maybe only in the newly released [ROS Melodic](https://discourse.ros.org/t/ros-melodic-morenia-release/4871/2)?)

@jlgarcia75 What is the best way to follow up on this ? open a new issue ? ask to reopen this one ?