realsenseai/librealsense #572: Debian package

vidstige, 2017-08-19:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |

For all us who use Debian, as well as derivates such as Ubuntu, it would be awesome to package this as a debian package. I'm thinking step one can be to just have a way of building a deb package for easy distribution. Step two can be to make a proper debian package and submit to the Debian repository for that super easy `sudo apt install librealsense` experience. Any thoughts on this?

I'd gladly make a pr for step one, and could possibly also do step two.

dorodnic, 2017-08-19:
Hi @vidstige 
At the moment, there are two ways to get **librealsense** as a binary package:
1. Using ROS PPAs with [ros-realsense](http://wiki.ros.org/RealSense)
2. Using [Intel managed PPA](https://software.intel.com/sites/products/realsense/intro/getting_started.html)
In addition, dedicated PPA will be provided with the launch of RS400 series this September. 
We are still considering the benefits of using standard repositories.
We can open-source the scripts we use for building the package, but these should probably reside somewhere else (as a separate repository perhaps)

vidstige, 2017-08-19:
Alright, thanks for your reply. I'll try with the Intel managed PPA first, that looks very promising.

vidstige, 2017-08-21:
How is the deb build for the Intel managed PPA? I see no way to build the deb in this repository?

vidstige, 2017-08-21:
The repo at http://realsense-alm-public.s3.amazonaws.com/apt-repo does not seem to be signed. My apt refuses to install these and refers to [apt-secure(8)](https://manpages.debian.org/jessie/apt/apt-secure.8.en.html).

dorodnic, 2017-08-21:
@vidstige 
Are you referring to the following warning:
`W: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/Release.gpg: Signature by key 5ED996B383636C252B681574AE3368787DCC6F0F uses weak digest algorithm (SHA1)`
While this is definitely fault of the repository, it should not prevent you from running:
`sudo apt-get install librealsense1`
To be on the safe side, just tested with clean Ubuntu 16.04.3 image. 
We should however improve our signing process for the future. 
In your previous comment, do you mean that you want to build the DEB file yourself?

vidstige, 2017-08-22:
No, I got a warning that I think happens when the `Release.gpg` file is missing. Can try again later today. I solved that by uploading `librealsense1` and `librealsense-dev` to our PPA instead.

I would like to avoid having to build it myself. But as we currently use a different version that what is available in the repo, I might have to. However, it seems we _can_ just upgrade to the version in your repo, so that's awesome.

vidstige, 2017-08-22:
Discard the above warning. Now it can find the gpg file, was probably just some network error or I was confused somehow.

vidstige, 2017-08-22:
Basically I would like to be able to generate a new debian package in case we want to upgarde, etc and the versions in the PPA is behind, etc. The lib seems pretty stable now though.

dorodnic, 2017-08-22:
I will check how we can share the scripts we are using.

closed 2017-11-01 by jlgarcia75