realsenseai/librealsense #573: LibRealSense vs Win-DCM-RealSense distortion model

avivhu, 2017-08-19:
We are using SR300.

The distortion parameters returned by librealsense's rs_get_stream_intrinsics() ("Inverse Brown Conrady" model) are different from the parameters returned by the Windows DCM driver. They are not the same, nor do the DCM parameters match the "Forward Brown Conrady" parameters.

I suppose this is intentional and they use different distortion models? This was implied by **Wollimayer** in [Issue 46](https://github.com/IntelRealSense/librealsense/issues/46).

Is there a way to retrieve the parameters of the Windows DCM distortion model using librealsense? Or maybe there's a known function that translates between the models? 
We want to substitute code that uses these parameters and that depends on the DCM with librealsense.

**Note 1:** To the best of our knowledge, the DCM distortion model is the same as the OpenCV distortion model, which is also the same as the Matlab Computer Vision Toolbox distortion model. Seems like someone might have dealt with these parameter translations before.

**Note 2:** We do get the same 3x3 projection matrix in both APIs.

**Note 3:** We tested with two SR300, one is an older development model and one is a commercial model from Creative.

| Required Info  |   |
|---|---|
| Camera Model |  SR300  | 
| Firmware Version |  3.21.0.0 | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |

oordan, 2017-08-21:
Which DCM API are you using? Is it Intel::RealSense::Calibration SDK interface?

avivhu, 2017-08-23:
@oordan  Yes, Intel::RealSense::Calibration::QueryStreamProjectionParameters()

oordan, 2017-08-29:
The intrinsic parameters are calculated the same for depth.
For color librealsense adjusts the focal length and principal point in case of 4:3 aspect ratio, which is not done in DCM, and the rest is the same.
The distortion parameters are the same for both libraries, only that librealsense provides them as: 
`float coeffs[5]`
and DCM provides them as: 
`tangential distortion = {coeffs[2], coeffs[3]}` 
`radial distortion = {coeffs[0], coeefs[1], coeffs[4]}`
but the data is supposed to be the same.
If you don't use 4:3 aspect ratio, do you get different numbers?

avivhu, 2017-08-29:
@oordan We are comparing the same depth stream with the same ratio 4:3. We get different values for the distortion coefficients.

We don't use color in our application.

We prepared sample code [here](https://1drv.ms/u/s!Alw927whRtzqkxVdytTZ7wmc-vWF). Compile it on Windows 10.

The LibRealSenseOutput is:
```
{fx, fy} = { 475.983948, 475.983826 }
{ppx, ppy} = { 313.900299, 245.987778 }
tangential distortion = { 0.004026, 0.004422 }
radial distortion = { 0.163121, -0.040345, 0.227614 }
```

The DCM output is:
```
{fx,fy} = { 475.9839, 475.9838 }
{ppx,ppy} = { 313.9003, 245.9878 }
tangential distortion = { -0.002873175, -0.002943313 }
radial distortion = { -0.1494821, 0.01267229, -0.02257685 }
```

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

avivhu, 2017-10-08:
We do still need help with this issue. We wrote code demonstrating it and but the developers haven't confirmed that it is an issue. 

I suppose librealsense devs can say it is not in their scope to deal with the Windows Realsense SDK compatibility?

We'll try the RealSense community but I think the persons in Intel who wrote the intrinsics code for RealSense SDK or for librealsense would have the best chance to answer our questions.

oordan, 2017-10-08:
We wrote the code for the Windows SDK, and from the code perspective - the numbers should be the same. So we need to debug your code to understand the diff. I'm sorry for the delay in handling this debug.

avivhu, 2017-10-10:
@oordan  Thanks for looking into this. We want to transfer our system from the Windows SDK to librealsense and the parameters mismatch is an issue we need to solve.

zivsha, 2018-02-03:
@avivhu were you able to resolve this issue?

avivhu, 2018-02-05:
No but I'm no longer working on this project.

closed 2018-02-12 by dorodnic