realsenseai/librealsense #574: R200: No matching media type for pixel format 494e565a

VanessaLange, 2017-08-21:
The latest commit on the master branch, which fixed issue #379 (commit # e5005d9), broke the R200 camera for me. The error message is:
no matching media type for pixel format 494e565a

| Required Info  |   |
|---|---|
| Camera Model | R200  | 
| Operating System & Version | Windows 10  |
| Build System | Visual Studio 2015 |

dorodnic, 2017-08-21:
@VanessaLange you are absolutely correct, we are working to fix it. Based on limited validation the issue should be resolved already, but we will double check this time

ev-mp, 2017-08-21:
Hi @VanessaLange , the [hotfix](https://github.com/IntelRealSense/librealsense/pull/575) is in place. 
Please get the latest version and update.

VanessaLange, 2017-08-22:
Thank you for fixing the issue so quickly! R200 and SR300 (issue #379) both work fine again.

ev-mp, 2017-08-22:
Closed with [PR575](https://github.com/IntelRealSense/librealsense/pull/575)

closed 2017-08-22 by ev-mp