realsenseai/librealsense #577: Question about Video4Linux backend preparation

riskeverything, 2017-08-23:
| Camera Model | SR300  | 
| Firmware Version |   | 
| Operating System & Version |Ubuntu 14.01|
| Kernel Version (Linux Only)  |4.4.0-92-generic|

When I installed Video4Linux backend preparation,some errors happened.
"./scripts/patch-realsense-ubuntu-xenial.sh",after that,It shows:

ls: cannot access /dev/video*: No such file or directory
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version.
git is already the newest version.
linux-headers-generic is already the newest version.
The following packages were automatically installed and are no longer required:
  libllvm3.6 libntdb1 linux-headers-4.2.0-27 linux-headers-4.2.0-27-generic
  linux-image-4.2.0-27-generic linux-image-extra-4.2.0-27-generic python-ntdb
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Applying realsense-uvc patch
can't find file to patch at input line 21
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 2fb29ac45b298b1dbea7fca174b6e8b7d22e4fb3 Mon Sep 17 00:00:00 2001
|From: Daniel Patrick Johnson <teknotus@teknot.us>
|From: Elad Zucker <eladz@hexsys-solutions.com>
|From: Evgeni Raikhel <eraikhel@intel.com>
|Date: Sun, 4 Jun 2017 17:30:59 +0300
|Subject: [PATCH] Add video formats for Intel® RealSense™ devices
|
|Signed-off-by: Evgeni Raikhel <eraikhel@intel.com>
|---
| drivers/media/usb/uvc/Makefile       |  1 +
| drivers/media/usb/uvc/uvc_driver.c   | 55 ++++++++++++++++++++++++++++++++++++
| drivers/media/usb/uvc/uvcvideo.h     | 35 ++++++++++++++++++++++-
| drivers/media/v4l2-core/v4l2-ioctl.c |  5 ++++
| include/uapi/linux/videodev2.h       |  6 ++++
| 5 files changed, 101 insertions(+), 1 deletion(-)
|
|diff --git a/drivers/media/usb/uvc/Makefile b/drivers/media/usb/uvc/Makefile
|index c26d12f..d86cf22 100644
|--- a/drivers/media/usb/uvc/Makefile
|+++ b/drivers/media/usb/uvc/Makefile
--------------------------
File to patch:

"File to patch",it lets me to fill in something?the previous orders didn't have errors.
And I can see the RGB image in rviz,but no RGB-D image.

ev-mp, 2017-08-23:
Hi @riskeverything, you're running stack 4.2 kernel. You need to upgrade it to 4.4 to realign with the LTS rolling kernels Ubuntu introduces since then. 
We provide Ubuntu patches for 14.04, 16.04.1 and 16.04.2 LTS with rolling kernels.
Follow the librealsense [linux installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#make-ubuntu-up-to-date) to upgrade the distribution, and then reapply the patch.

See more on LTS enablement on [Ubuntu site](https://wiki.ubuntu.com/Kernel/LTSEnablementStack#Ubuntu_14.04_LTS_-_Trusty_Tahr)

riskeverything, 2017-08-23:
But,I use "uname -r",It shows "4.4.0-92-generic"

ev-mp, 2017-08-23:
What is the command line you use to invoke the patch?

riskeverything, 2017-08-23:
"./scripts/patch-realsense-ubuntu-xenial.sh" in the folder named librealsense-master.

ev-mp, 2017-08-23:
You need to check whether "git clone" was able to download the kernel sources:
Go into "ubuntu-xenial-master" directory that should be present inside "librealsense-master" folder:
 - Check the kernel version with "git log -1"
 - Browse deeper into "ubuntu-xenial-master/drivers/media/usb/uvc" and make sure that the source files (including Makefile) are present.

Ultimately, modify and rerun "patch-realsense-ubuntu-xenial.sh" and replace "set -e" with "set -ex" in line 4, then post the log

riskeverything, 2017-08-23:
Maybe the questions is about git.I use "git log -1" in folder named "ubuntu-xenial-master",it shows fatal: bad default revision 'HEAD'.And the folder named ubuntu-xenial-master has nothing in it.I test my "git clone" to clone the "librealsense-master" ,it is slow but working.

ev-mp, 2017-08-23:
In case the local repo is broken, then deleting "ubuntu-xenial-master" and verifying git status should suffice.
But more often this is a firewall/proxy issue that prevents accessing git launchpad and that needs to be resolved by some 3rd party. In that case you can run the script on a non-protected PC to download kernel sources, then manually transfer it to the original PC.

riskeverything, 2017-08-24:
OK,Thanks.Yesterday，I test many times to install it.I retboot my PC many times,but it doesn't work.Today, I remove the folder named ubuntu-xenial-master,and I use patch again.It is working.That's amazing.Thank you again.

closed 2017-08-24 by riskeverything