realsenseai/librealsense #578: Only depth stream works well

nanowhiter, 2017-08-25:
Hi, 
I am working on a Project with Python 3.5. 
I can get the depth image from R200, but not RGB, and any other images.
The machine is connecting and the function dev.wait_for_frames() is working but the values in dev.color are all zero. 
One more weird thing is that the light on R200 is on when I only use DepthStream. 
I am pretty sure that the RGB camera is fine because I tested it with C++ and it worked well.

Thank you,

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.10 | 
| Operating System & Version |  Windwos 10 |

ev-mp, 2017-08-27:
Hi @nanowhiter ,
It is not clear whether the issue is describe is related to Python wrapper, or Librealsense core. 
What Python wrapper you're running with?  What do you mean by  _I tested it with C++ and it worked well_ ?

More questions/hints:
 - What happens when you configure `cpp-config-ui` to depth/ir/color only, does it produce any data ?
 - Can you run `cpp-config-ui` and `cpp-headless` and describe/share the results.
 -  Additionally, you can verify RGB sensor with a Microsoft tool called GraphEdit (look up [this thread](https://github.com/IntelRealSense/librealsense/issues/481)), and at the same time to check whether the Microsoft UVC driver is properly installed.

With regards to the `light` - you probably refer to the light emitter which is normally turned on when the depth stream is activated.

nanowhiter, 2017-09-18:
Hi, @ev-mp ,
Thank you for your replying. "cpp-config-ui" and "cpp-headless" have the same problem: no matching media type for  pixel format 59555932. And the solution by  @hinmanj worked for me. 
Thanks a lot,
#580

closed 2017-09-18 by nanowhiter