realsenseai/librealsense #583: It works with linux, but can it work with ARM processors such as on the Raspberry Pi?

AnthonyTheKoala, 2017-08-27:
I would like a question that is not addressed by the SDK's main page.
I understand that the SDK will work in Linux machines.
However, will it work with Raspberry Pi which is an ARM processor and USB 2.0?

Another question not addressed on the SDK's main page. I want to use the functions of opencv. Has opencv been used with the real sense cameras.

Thank you,
Anthony from Sydney Australia

| Camera Model | R200 / F200 / SR300 / ZR300 | 
| Operating System & Version | Linux (Ubuntu) |

dorodnic, 2017-08-27:
Hello @AnthonyTheKoala 
So, there are couple of points to address:
1. Intel RealSense Cameras **require** USB3. This is due to large bandwidth required to pass both depth and color data from the device. In theory if you are satisfied with low resolutions it can be done, but it would require dedicated firmware support and is not part of any RealSense product currently on the market. 
2. While we do not **officially** support ARM, librealsense is an open source project and it has been shown to work on ARM based devices (NVidia Jetson TX2 is the most appropriate alternative)
3. Color stream unpacking and point-cloud generation can be quite CPU intensive, so even on the Jetson it is recommended to offload computation to the GPU. I am not sure Raspberry Pi could handle it. 
4. We offer a very basic [tutorial](https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md) for working with OpenCV.
5. There are many small form-factor alternatives, please see [hardware-requirements](https://github.com/IntelRealSense/librealsense#hardware-requirements). The [UP Board](http://www.up-board.org/up/kickstarter/up-intel-realsense-technology/) in particular is something you might want to check - it shares the form factor with the Raspberry Pi and works very well.

closed 2017-09-11 by dorodnic

nitishwadne, 2019-02-23:
can we use intel realsense D435 with nano pi M4 which is arm architecture. where can i get the installation setup like we have different steps for raspberry pi and different steps for nvidia jetson.