realsenseai/librealsense #587: Google chrome, Chromium, Firefox - not working gesture?

shamun, 2017-08-29:
Google chrome, Chromium, Firefox - not working gesture? How to natively get gesture and informations to browser?

freemanlo, 2017-09-12:
What's your test configuration? 
1) Win10 RS1? SR300 or F200?
2) which DCM version did you install?
3) Is the RealSense device showing in the device manager correctly?

jlgarcia75, 2017-09-12:
The web functionality of RealSense was discontinued several months ago due to a security issue with some of the dependencies. See this article for more details https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00066&languageid=en-fr. This means that the Javascript components for RealSense are no longer functional.

closed 2017-09-12 by jlgarcia75

halton, 2017-09-27:
@jlgarcia75 this repo is for librealsense and wrappers, the chrome browser support is happening different location.

You need report to https://bugs.chromium.org, Blink>GetUserMedia or Blink>MediaStream, your right owner is aleksandar.stojiljkovic@intel.com 

For more information about RealSense on browser, please check https://software.intel.com/en-us/blogs/2017/06/22/tutorial-typing-in-the-air-using-depth-camera-chrome-javascript-and-webgl-transform

shamun, 2017-09-27:
> The web functionality of RealSense was discontinued several months ago due to a security issue with some of the dependencies. See this article for more details https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00066&languageid=en-fr. This means that the Javascript components for RealSense are no longer functional.
> 
> Jesus
> Intel Customer Support

We need that feature back. Its pain to have it removed and now its nightmare to get it available in browser. One has to spend day and night coding and adjusting C#, C++, QT5, Java all those pain was not in old implementation. It was simplified by adding the Javascript library and work directly in the browser.

Can we have that feature back in new hardware or existing hardwares once again?
(i still have frozen Intel Javascript in OLD system they works even Intel removed it, i clone it and deepfreeze that system. But when i go to new installation it does not work so i do clone of my old setup to new systems now). For me security is not important because i run the system in my own management. Therefore removing such feature completely from the Intel is to me like nonsense.

shamun, 2017-09-27:
@halton: In Sep, 2017. 
Is it working with Chromium and Javascript? Or still same issue as mentioned they removed it: https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00066&languageid=en-fr

halton, 2017-09-27:
@shamun I'm 100% sure the depth camera  (realsense is included) support and developing still happening on Chromium.

shamun, 2017-09-27:
@halton: that will be a live saving. 

Do you have by chance, follow up reference article from official Intel (stating NodeJS or Javascript library or Chromium progress or road map)?

halton, 2017-09-27:
@shamun 

Node.js support is happening on this repo, now is in the `development` branch, and [node-librealsense](https://www.npmjs.com/package/node-librealsense) on npmjs.com is keeping to updated.

For the Chrome browser, it is totally different, librealsense is not used there, the API is following the [W3C Media Capture Depth Stream Extensions](https://www.w3.org/TR/mediacapture-depth/) .