realsenseai/librealsense #591: Camera could not be detected even after successful driver installation

sohamm17, 2017-08-31:
All my drivers are properly installed but the camera could not be detected and examples could not be run.
Getting the following error in `dmesg`:

```
[651750.832215] usb 3-2.1: new SuperSpeed USB device number 7 using xhci_hcd
[651750.848998] usb 3-2.1: New USB device found, idVendor=8086, idProduct=0aa5
[651750.849003] usb 3-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[651750.849005] usb 3-2.1: Product: Intel(R) RealSense(TM) Camera SR300
[651750.849007] usb 3-2.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[651750.850473] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[651750.852284] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.0/input/input57
[651750.852677] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[651750.852681] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[651750.852738] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[651750.854285] uvcvideo: Unable to create debugfs 3-7 directory.
[651750.854396] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.1/3-2.1:1.2/input/input58
[651751.458021] ------------[ cut here ]------------
[651751.458038] WARNING: CPU: 1 PID: 3437 at /build/linux-lts-xenial-ytXsCy/linux-lts-xenial-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd3e/0x10c0 [videodev]()
[651751.458041] Unknown pixelformat 0x2036315a
[651751.458042] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media snd_usb_audio snd_usbmidi_lib hid_generic rt2800usb rt2x00usb rt2800lib rt2x00lib crc_ccitt btrfs xor raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c rndis_host cdc_ether usbnet nls_utf8 isofs uas usb_storage drbg ansi_cprng ctr ccm 8192cu(OE) rtl8xxxu rtl8812au(OE) msr pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) cmac bnep rfcomm binfmt_misc rtsx_usb_ms memstick hid_multitouch btusb btrtl btbcm btintel bluetooth arc4 dell_led intel_rapl iwlmvm x86_pkg_temp_thermal intel_powerclamp coretemp snd_hda_codec_hdmi kvm_intel snd_hda_codec_realtek snd_hda_codec_generic kvm snd_soc_rt5640 snd_hda_intel snd_hda_codec snd_soc_rl6231 irqbypass mac80211 snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine snd_hda_core crct10dif_pclmul i915 snd_hwdep crc32_pclmul ghash_clmulni_intel snd_pcm aesni_intel aes_x86_64 snd_seq_midi snd_seq_midi_event dell_laptop dcdbas lrw dell_smm_hwmon dell_wmi drm_kms_helper drm sparse_keymap iwlwifi snd_rawmidi hid_rmi gf128mul snd_seq glue_helper snd_seq_device i2c_algo_bit snd_timer ablk_helper joydev input_leds cryptd serio_raw elan_i2c cfg80211 fb_sys_fops syscopyarea snd sysfillrect sysimgblt soundcore i2c_hid wmi mei_me video parport_pc mei dw_dmac i2c_designware_platform spi_pxa2xx_platform shpchp dw_dmac_core lpc_ich mac_hid 8250_dw dell_rbtn ppdev i2c_designware_core soc_button_array lp parport rtsx_usb_sdmmc rtsx_usb usbhid hid psmouse ahci r8169 libahci mii sdhci_acpi sdhci fjes [last unloaded: videobuf2_memops]
[651751.458147] CPU: 1 PID: 3437 Comm: AudioThread Tainted: G        W  OE   4.4.0-92-generic #115~14.04.1-Ubuntu
[651751.458148] Hardware name: Dell Inc. Inspiron 5547/0598GM, BIOS A04 05/09/2014
[651751.458149]  0000000000000000 ffff88007d15fbe8 ffffffff813de5bc ffff88007d15fc30
[651751.458152]  ffffffffc0801ab0 ffff88007d15fc20 ffffffff8107ea36 ffff88007d15fd98
[651751.458154]  0000000000000000 0000000000000000 ffff8802243aec00 ffffffffc0804180
[651751.458156] Call Trace:
[651751.458160]  [<ffffffff813de5bc>] dump_stack+0x63/0x87
[651751.458164]  [<ffffffff8107ea36>] warn_slowpath_common+0x86/0xc0
[651751.458167]  [<ffffffff8107eabc>] warn_slowpath_fmt+0x4c/0x50
[651751.458171]  [<ffffffffc0ef3187>] ? uvc_ioctl_enum_fmt+0xb7/0x140 [uvcvideo]
[651751.458177]  [<ffffffffc07eb80e>] v4l_enum_fmt+0xd3e/0x10c0 [videodev]
[651751.458181]  [<ffffffffc07ebdea>] __video_do_ioctl+0x25a/0x2e0 [videodev]
[651751.458184]  [<ffffffff812212f4>] ? mntput+0x24/0x40
[651751.458187]  [<ffffffff8120f319>] ? path_openat+0x2e9/0x12d0
[651751.458191]  [<ffffffffc07ee3fb>] video_usercopy+0x21b/0x590 [videodev]
[651751.458195]  [<ffffffffc07ebb90>] ? v4l_enum_fmt+0x10c0/0x10c0 [videodev]
[651751.458198]  [<ffffffff81211682>] ? do_filp_open+0x92/0xd0
[651751.458202]  [<ffffffffc07ee785>] video_ioctl2+0x15/0x20 [videodev]
[651751.458206]  [<ffffffffc07e7644>] v4l2_ioctl+0xa4/0xc0 [videodev]
[651751.458209]  [<ffffffff8121474d>] do_vfs_ioctl+0x2dd/0x4c0
[651751.458210]  [<ffffffff81210763>] ? putname+0x53/0x60
[651751.458212]  [<ffffffff81200828>] ? do_sys_open+0x1b8/0x270
[651751.458215]  [<ffffffff812149a9>] SyS_ioctl+0x79/0x90
[651751.458217]  [<ffffffff8180cd36>] entry_SYSCALL_64_fastpath+0x16/0x75
[651751.458219] ---[ end trace 0c739070e5435182 ]---
[651751.458325] ------------[ cut here ]------------
```

| Required Info  |   |
|---|---|
| Camera Model | SR300  | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  |  4.4. |

dorodnic, 2017-08-31:
Hello @sohamm17 
Do you see the device when running `lsusb`? 
What error are you getting when you run the samples?
Did the [Kernel patch](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) required for SR300 pixel formats applied successfully?

sohamm17, 2017-08-31:
Yes, with lsusb I believe it's detecting the device:
`Bus 003 Device 015: ID 8086:0aa5 Intel Corp.`
When I plug off the device, this entry is no longer there.
I have patched the kernel and it says that the `uvcdriver` is registered.
When I run the examples, it says no device connected.

sohamm17, 2017-08-31:
I think I have figured out the problem. I was trying it on my laptop which had another integrated webcam. I have disabled that webcam following this [thread](https://askubuntu.com/questions/189708/how-to-disable-integrated-webcam-and-still-be-able-to-use-an-external-one).

closed 2017-08-31 by sohamm17