realsenseai/librealsense #592: Multiple ZR300 indoor settings

jgomezlopez, 2017-09-05:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.
I'm trying to get Depth images with 3 x ZR300  cameras. I see some interferences between them. What are the right settings to capture depth images without interferences in a indoor environment?
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   | 
| Operating System & Version | ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.9  |

dorodnic, 2017-09-05:
Hi @jgomezlopez 
Could you please share what kind of artifacts are you seeing? (You can attach images to GitHub comments)
In theory ZR300 laser interference is not supposed to be destructive. In practice, LR auto-exposure, as well as exposure and gain values can have effect on the output.

jlgarcia75, 2017-09-26:
Hello @jgomezlopez , do you still need help with this issue? As @dorodnic mentioned, there may be some artifacts visible if the cameras IR lasers are pointed at each other but it can be managed by adjusting the exposure and gain values.

Jesus
Intel Customer Support

jlgarcia75, 2017-09-29:
Closing case. Status: Abandoned by Customer.

Jesus
Intel Customer Support

closed 2017-09-29 by jlgarcia75