realsenseai/librealsense #605: Support for D415 and D435 cameras?

severin-lemaignan, 2017-09-14:
The [D415 and D435 product page](https://software.intel.com/en-us/realsense/400-series) mentions an open-source SDK. I assume the cameras will be supported by `librealsense`, but I see no mention of it so far. Can you confirm these new cameras will be supported in a cross-platform way through `librealsense`?

severin-lemaignan, 2017-09-14:
Ok, I answer my own question: support is indeed planned in `librealsense`, in the [`development` branch](https://github.com/IntelRealSense/librealsense/tree/development).

closed 2017-09-14 by severin-lemaignan

nouei, 2018-01-15:
Yes I it works with librealsense. Just needs to follow this instruction and make it after.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md