realsenseai/librealsense #632: Bogus "Requires.private: Threads" in realsense2.pc

morxa, 2017-09-21:
I recently updated librealsense to the 2.7.9 alpha release in Fedora Rawhide [1]. The package builds fine, but it generates a bogus dependency in the pkg-config file `realsense2.pc`:

```
prefix=/usr
exec_prefix=${prefix}
includedir=${prefix}/include
#TODO: libdir=${exec_prefix}/lib
libdir= ${prefix}/lib/x86_64-linux-gnu

Name:
Description: Intel(R) RealSense(tm) Cross Platform API
Version: 2.7.9
URL: https://github.com/IntelRealSense/librealsense
Requires.private:  Threads libusb-1.0
Libs: -L${libdir} -lrealsense2
Libs.private: 
Cflags: -I${includedir}

#TODO check -Wl -Bdynamic
#Libs: -L${libdir} -Wl,-Bdynamic -lrealsense
```

The problem is with this line:

```
Requires.private:  Threads libusb-1.0
```

At least in Fedora, there is no `Threads.pc`, i.e., no package provides Threads. Thus, using (and in fact installing) the package is currently not possible, because of the unresolved pkg-config requirement.

I don't know about other distributions, so I'm not sure what to suggest. Is there a Threads.pc in Debian/Ubuntu? I don't have access to any Debian-based system, so I can't check, but I would expect the dependency to be broken on all distributions. If that's the case, removing [line 458 in CMakeLists.txt](https://github.com/IntelRealSense/librealsense/blob/development/CMakeLists.txt#L458) should fix the problem.

Otherwise, can we introduce a case distinction or check if something provides Threads.pc before adding the dependency? I'd like to avoid a downstream patch.

[1] https://src.fedoraproject.org/rpms/librealsense

jlgarcia75, 2017-10-03:
Hello @morxa, 

Closing this issue due to fix reported here: https://src.fedoraproject.org/rpms/librealsense/c/afe6b721f0e84763c07184d58f8f9d180ceae802.

Otherwise, if the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75

morxa, 2017-10-03:
@jlgarcia75 thanks for the comment. Well, the fix is actually mine, as I've mentioned above :) The purpose of this report was to figure out whether this requirement is intended. At Fedora, we try to keep the number of downstream patches minimal and upstream most fixes.

That being said, I'm not sure I understand your comment correctly. I thought librealsense2 is the current development version? 

I just realized that librealsense2 does not support the older cameras. As a Fedora package maintainer, what do you recommend me to do? Should I have two separate packages, one for the legacy release and one for the new librealsense2?

jlgarcia75, 2017-10-03:
Hi @morxa,

I actually don't know if that requirement is intended. Maybe @dorodnic can comment on that.

To answer your question on librealsense2 vs librealsense1...Librealsense 2 is currently under development and will only support the new D400 series cameras and the SR300 camera. Librealsense 2 will not support R200 or ZR300. Librealsense 1 is no longer being worked on so any bugs relating to it will not be fixed.  Does this help?

Regards,
Jesus
Intel Customer Support

jlgarcia75, 2017-10-03:
@morxa, I forgot to mention that the R200 and ZR300 cameras are discontinued so no bugs relating to hardware, firmware, or software will be fixed.

morxa, 2017-10-05:
@jlgarcia75 yes, thank you for your comments. This means I will revert librealsense to the legacy release and submit a new librealsense2 to Fedora. I guess that's how Debian does it, too.

morxa, 2017-12-13:
@dorodnic Can you comment on the original question, the dependency on `Threads`?

BjarneHerland, 2018-01-17:
I created pull-request #1026 since this is pretty annoying when frequently rebuilding the SDK.