realsenseai/librealsense #656: SR300 - Very noisy depth data

AndyZe, 2017-09-26:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | current camera firmware is 3.15.0.0  | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  | 4.10.0-35-generic  |

Installed by "sudo apt install ros-kinetic-realsense-camera"

I appreciate the improvements you've been making to camera packaging and what seems like more robust USB connections. However, the depth data I'm seeing is basically unusable. Here's a forward-facing depth image of my hand. It looks OK-ish.

[https://ibb.co/bTU9ik](https://ibb.co/bTU9ik)

But from the side view, you can see the point cloud is awful.
[https://ibb.co/hB9LpQ](https://ibb.co/hB9LpQ)

AndyZe, 2017-09-26:
I think this is an issue with the ROS wrapper. When I manually install librealsense and run cpp_tutorial_2_streams, the depth data looks very good. See [https://ibb.co/dWcOR5](https://ibb.co/dWcOR5)

AndyZe, 2017-09-26:
Fixed by setting these lines in sr300_nodelet_default.launch:

```
  <arg name="hw_registered_processing"        default="true" />
  <arg name="sw_registered_processing"        default="false" />
```

closed 2017-09-26 by AndyZe

AndyZe, 2017-09-26:
I should also mention, I gave up  on the debian installation (sudo apt install ros-kinetic-realsense-camera) and installed this librealsense package from source. I guess that was the real solution, not the switch to hw_registered_processing.

tolgasaglik, 2018-05-08:
I use the sr300_nodelet_rgbd.launch file and am able to retrieve 3d information in terms of colors. The rgb info of dept is puslished in /camera/depth/points topic. After that point, I am not sure how to extract information from it. Do you have any progress on obtaining it?