realsenseai/librealsense #659: R200 not recognized under Linux Mint, multiple "Unknown pixelformat"-errors despite patched kernel

mtee, 2017-09-27:
So after following the installation notes [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) 
including doing
`./scripts/patch-realsense-ubuntu-xenial.sh`

> Compiling uvc module
> make: Entering directory '/mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1'
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_driver.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_queue.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_v4l2.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_video.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_status.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_ctrl.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_isight.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_entity.o
>   CC [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvc_debugfs.o
>   LD [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvcvideo.o
>   Building modules, stage 2.
>   MODPOST 1 modules
>   CC      /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvcvideo.mod.o
>   LD [M]  /mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1/drivers/media/usb/uvc//uvcvideo.ko
> make: Leaving directory '/mypath/librealsense/ubuntu-xenial-Ubuntu-hwe-4.8.0-58.63_16.04.1'
> Patched kernel module created successfully
> 
> Handle uvcvideo :
> Module is resident, unloading ...  succeeded. 
> Applying the patched module ...  succeeded
> 
> 
> Script has completed successfully. Please consult the installation guide for further instruction.

and sudo modprobe uvcvideo

I still cannot get R200 working. Every time I reconnect the camera, the dmesg gets these 3 "unknown pixelformat" messages.
At first, the camera seems to be successfully recognized:

> [  228.047802] usb 2-2.1: new SuperSpeed USB device number 6 using xhci_hcd
> [  228.070445] usb 2-2.1: New USB device found, idVendor=8086, idProduct=0a80
> [  228.070449] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  228.070453] usb 2-2.1: Product: Intel RealSense 3D Camera R200
> [  228.070455] usb 2-2.1: Manufacturer: Intel Corp
> [  228.070457] usb 2-2.1: SerialNumber: SN_2521002335
> [  228.072775] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [  228.076235] uvcvideo 2-2.1:1.0: Entity type for entity Extension 2 was not initialized!
> [  228.076241] uvcvideo 2-2.1:1.0: Entity type for entity Camera 1 was not initialized!
> [  228.077065] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [  228.080147] uvcvideo: Unable to create debugfs 2-6 directory.
> [  228.080348] uvcvideo 2-2.1:1.2: Entity type for entity Camera 4 was not initialized!
> [  228.080595] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1:1.2/input/input25
> [  228.081575] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [  228.084756] uvcvideo: Unable to create debugfs 2-6 directory.
> [  228.084943] uvcvideo 2-2.1:1.4: Entity type for entity Processing 7 was not initialized!
> [  228.084949] uvcvideo 2-2.1:1.4: Entity type for entity Camera 6 was not initialized!
> [  228.085231] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.1/2-2.1:1.4/input/input26
> [  228.101508] ------------[ cut here ]------------
> [  228.101516] WARNING: CPU: 0 PID: 3093 at /build/linux-hwe-0IbwuF/linux-hwe-4.8.0/drivers/media/v4l2-core/v4l2-ioctl.c:1286 v4l_enum_fmt+0xe24/0x1190 [videodev]
but then i get
```
Unknown pixelformat 0x49323159
Unknown pixelformat 0x20493859
Unknown pixelformat 0x2036315a

```
with stacktraces inbetween.
Does it have to do with using Linux Mint instead of Ubuntu or some incompatible kernel version perhaps?

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.10  | 
| Operating System & Version | Linux Mint 18.2 64bit  |
| Kernel Version (Linux Only)  | 4.8.0-58-generic  |

ev-mp, 2017-09-27:
Hi @mtee, can you be more specific of what librealsense demo/app did not run correctly ?

As for the log messages you refer to are **warnings** that and are produced by another [videodev] kernel module, and they are not indicative of patch success or failure 
> [ 228.101516] **WARNING**: CPU: 0 PID: 3093 at /build/linux-hwe-0IbwuF/linux-hwe-4.8.0/drivers/media/v4l2-core/v4l2-ioctl.c:1286 v4l_enum_fmt+0xe24/0x1190 [videodev].

Actually, looking into the log you posted both the patching process and R200 enumeration of R200 completed successfully, 
In order check whether the required formats have been resolved you can run the following (requires 'v4l-utils' installed):

_v4l2-ctl --list-formats-ext -d /dev/video0
v4l2-ctl --list-formats-ext -d /dev/video1
v4l2-ctl --list-formats-ext -d /dev/video2_

and check whether there are unrecognized formats.

Moreover, in case you're using R200 only you may skip the patch completely since its format descriptions have already been upstreamed to Linux kernel since version 4.4.+.

mtee, 2017-09-28:
Hi, @ev-mp, thank you for your reply. After all, the unrecognized R200 was likely a hardware or a firmware defect. I have two R200s here, and one of them is recognized fine. However, it is also affected by the sudden crashes while running examples.
In particular, after I start 
librealsense/build $ examples/cpp-tutorial-1-depth
or 
librealsense/build $ examples/cpp-tutorial-2-streams
it runs for a few seconds and then the image freezes, the following shows up in the terminal:

> rs::error was thrown when calling rs_wait_for_frames(device:0x262fd40):
>     Timeout waiting for frames.

and there is also a

> Non-zero status (-71) in video completion handler

popping up in the dmesg when the crash happens.

If I try to restart the example, I get

> rs::error was thrown when calling rs_create_context(api_version:11203):
>     UVCIOC_CTRL_QUERY:UVC_SET_CUR error 2, No such file or directory

in the terminal, accompanied by

> uvcvideo: Failed to query (GET_LEN) UVC control 1 on unit 2: -71 (exp. 2).

in the dmesg.
Same happens with all other examples:

> examples/cpp-enumerate-devices 
> RealSense error calling rs_create_context(api_version:11203):
>     UVCIOC_CTRL_QUERY:UVC_SET_CUR error 2, No such file or directory

To circumvent the error I either have to reboot, or disconnect the camera, reconnect it and wait for a while before starting the example, otherwise it won't start and will show the same error as above.
In the meantime I have also installed the kernel 4.11, it sadly had no effect on the issue whatsoever. Hooking up the camera to a Win10 machine also didn't help. The DCM detects it just fine, does not ask for a firmware update though.

ev-mp, 2017-09-28:
Hi @mtee ,
There are indications of power management issues ( suspend mode) .
Additionally, the intermittent errors that you describe also suggest that the HW setup is not reliable.

Can you post a complete dmesg log from connecting the camera till the application crash?

Did you try to switch USB port and/or connect R200 via self-powered USB3 hub?
I also recommend to try the device on a separate Linux host, preferably with Ubuntu to isolate the issue.

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello @mtee,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by ev-mp