realsenseai/librealsense #661: 410 support?

vidstige, 2017-09-28:
Hi,
I've got a Customer Engineering Sample of the mythical R410 in front of me. However, there seems to be no information included in the package on how to retrieve the depth stream. I pulled the latest sources and built librealsense from scratch, but still no cigar. Checking through the code the product id the hardware reports is not recognized by librealsense. Should I try to patch in the zr300 implementation for this new product id? In short: How do I get the depth stream from a R410?

| Required Info  |   |
|---|---|
| Camera Model | R410 | 
| Firmware Version |  ? | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.12.0-041200-generic  |

vidstige, 2017-09-29:
After reading through the fine print contained in the box I found the following url: platformsw.intel.com

However, it's protected by username password login, something which I do not have.

dorodnic, 2017-09-29:
Hi @vidstige 
Please try [the latest software release](https://github.com/IntelRealSense/librealsense/releases/tag/v2.7.9)
Let us know if you encounter any problem

vidstige, 2017-09-29:
worked perfectly, thank you. I was on latest master, you must use some kind of branching strategy that is beyond me. Anyway, thanks so much for the help. :bowing_man:

closed 2017-09-29 by vidstige

vidstige, 2017-09-29:
I need to run e.g. the capture binary with sudo or receive the following error

    RealSense error calling rs2_start_pipeline(pipe:0xaaacc0):
        libusb_open(...) returned LIBUSB_ERROR_ACCESS Last Error: Permission denied

There are two new device files added: `/dev/video0` and `/dev/video1`, both owned by group `video` and I'm myself member of video....

vidstige, 2017-09-29:
Never mind. Plopping the latest udev rules and triggering them solved it! :)

vidstige, 2017-09-29:
Ok, one more question: How do I apply the `short_range` preset? :thinking: 

In the rs-capture example I can do the following

    rs400::advanced_mode advanced(pipe.get_device());

But how do I actually apply a preset?

vidstige, 2017-09-29:
Tried to `serialize_json()` in hopes of then tweak all the parameters, and then `load_json()` but it fails with `Operation failed with error code=-6`...

oordan, 2017-10-01:
Can you please check your firmware version - using [realsense-viewer](https://github.com/IntelRealSense/librealsense/tree/development/tools/realsense-viewer) or [enumerate-devices](https://github.com/IntelRealSense/librealsense/tree/development/tools/enumerate-devices)?

vidstige, 2017-10-02:
Yeah, `rs-enumerate-devices` gives the following

```
 Device info: 
    Name                          : 	Intel RealSense 410
    Serial Number                 : 	701512020674
    Firmware Version              : 	05.05.07.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD2

```

vidstige, 2017-10-02:
@oordan my firmware is `05.05.07.00`. Updated the issue above. Do I need to update the firmware? Perhaps I should file a new issue for this. :thinking:

dorodnic, 2017-10-02:
As suspected the firmware is very old. I don't think this firmware supported presets (in particular advanced-mode commands)
The firmware update tool for the RS400 has not been published publicly yet. 
If you received the device from Intel representatives please contact them. I recommend asking for RS435 QS model replacement to ensure you get the most mature hardware, but RS415 should also become available soon and they are more similar to the 410 you have now.

vidstige, 2017-10-02:
Alright, I'll try to contact someone. I think we just got box delivered to our office. Which model/make would be best suited for us? We're using a rig with four of these cameras to scan feet. The feet are ~250-700mm away from camera. Accuracy is of essence so the higher resolution the better, more or less. Currently we're running at 640x480.

By recording at 640x480 I can get the 410 with the old firmware to register depths in our range, so that's good news. :-)

vidstige, 2017-10-03:
@lukasTallund what contacts at Intel do we have? Could you try to get an RS435 QS model replacement or RS415?