realsenseai/librealsense #662: Kernel patch on ARM to detect camera

Maxime59, 2017-09-28:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 | All
| Firmware Version |   | 
| Operating System & Version |   | Linux ARM distrib (Armbian, Linaro...)
| Kernel Version (Linux Only)  |   |

Hi everyone,

I'd like to use a ZR300 camera with an ARM-based embedded computer like the ASUS Tinker Board but even if I can compile the library and examples without any problem I can't get the camera to be detected.

So I have 2 questions : 
1) Is there any way I can succeed to detect the camera with such architecture ?
2) As those peripherals have a lot of applications into embedded systems (drones, robotics...) why that architecture is not represented ? 

Any help will be greatly appreciated ! :)

Best regards,

Max

dorodnic, 2017-09-29:
The key factor in this would be the exact Kernel version. Kernel patches that we provide are platform agnostic, but they are targeted at common versions used in Ubuntu distributions. 

NVidia Jetson TX2 is an ARM device with a custom kernel, and it can run RealSense ([more info](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_jetson.md))

Once you have the device recognized by the OS, there is also the fact that librealsense is optimized for Intel platforms ([this part](https://github.com/IntelRealSense/librealsense/blob/master/src/image.cpp)) From limited experience with the TX2, GPU acceleration would be almost required to do anything useful.  
You can check some of the work that was done by community members in the past ([example here](https://github.com/Maghoumi/culibrealsense))

Maxime59, 2017-09-29:
Thank you for your reply !

Actually I'm running a Linaro distrib built for the Tinker Board based on a Debian kernel.
The `uname -a` command returns : `Linux linaro-alip 4.4.16-00006-g4431f98-dirty #1 SMP Mon Apr 17 17:27:25 CST 2017 armv7l GNU/Linux`

So as far as I understand I have a kernel version that match the requirements of the patch but as it is not a common distrib the script doesn't find the sources of the kernel to apply the patch and re-build it is it right ?

dorodnic, 2017-10-02:
Right. You can take a look at the community-contributed [script for Fedora distribution](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-uvcvideo-fc25.sh) for reference. 
Also it is worth noting that if you have an option to upgrade to newer kernel version, the patches for RealSense has been upstreamed.

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello @Maxime59,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

Maxime59, 2017-10-09:
A patch for an ARM distrib kernel would be nice  :)

jlgarcia75, 2017-10-09:
Hello @Maxime59,

As you can imagine and have experienced, support for ARM architectures is virtually non-existent, and so far only limited to the Jetson TX2 board (and even that is minimal). RealSense cameras are developed for and validated on Intel Architecture-based platforms.

It is up to the community to provide patches for other architectures.

Regards,
Jesus
Intel Customer Support

closed 2017-10-12 by dorodnic