realsenseai/librealsense #665: R200 and ZR300 not working properly on new librealsense version

KappesR, 2017-09-28:
| Required Info  |   |
|---|---|
| Camera Model | R200 / ZR300 | 
| Firmware Version |  1.0.72.06 / 1.0.72.10  | 
| Operating System & Version |  Ubuntu 16.04.3 LTS  |
| Kernel Version (Linux Only)  |  4.10.0-35-Generic  |

I run in some trouble with the Intel RealSense R200 two weeks ago (the same problem came up for the ZR300 and another R200).
The R200 had the FW/V `1.0.72.06`.
The problem is that the two R200s and the ZR300 do not work in any way, neither with the librealsense library and demos nor with any other tools on Ubuntu like cheese, q4l2 and v4l2.
But I got always device information by lsusb -D /dev/bus/usb/...:
```
Device: ID 8086:0a80 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0a80 
  bcdDevice           14.8a
  iManufacturer           1 Intel Corp
  iProduct                2 Intel RealSense 3D Camera R200
  iSerial                 3 SN_2521001926
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength         1677
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              126mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               4 Intel(R) RealSense(TM) 3D Camera (R200) Left-Right
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              4 Intel(R) RealSense(TM) 3D Camera (R200) Left-Right
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.10
        wTotalLength           68
        dwClockFrequency       48.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
[... 354 lines cut ...]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/[user]/[catkin_ws]/src
ROS path [2]=/opt/ros/kinetic/share
```
But this may be a problem with the realsense repo.

I got older versions of realsense-repo with librealsense v1.11.0 and v1.11.1 to be running and can complie/build the newest version 2.0 of librealsense and realsense but the R200 is not working with them.

I also had the problem while FW update that the calibration tool returns an error DS_INVALID_ARG.

I think there is a problem with the librealsense library, but I had no suggestions for a further insight or to investigate this.

Strange is to solve this by a FW update.
The errors of the uvcvideo driver is more troublesome.

The stable versions for me are FW/V 1.72.0.10, librealsense v1.12.1, realsense 1.8.0.

dorodnic, 2017-09-29:
Hello @KappesR 
Please use version 1.12.1 with R200 / ZR300 models. 
Versions 2.x support only D400 and SR300 series. 
See supported devices under [releases](https://github.com/IntelRealSense/librealsense/releases)

KappesR, 2017-09-29:
Hi @dorodnic ,
thanks for the reponse!
There is may be a misunderstanding.
I already use v1.12.1 for R200 with realsense repo version 1.8.0:
> The stable versions for me are FW/V 1.72.0.10, librealsense v1.12.1, realsense 1.8.0.

But dmesg and `roslaunch realsense_camera r200_nodelet_rgbd.launch` throw still the above messages.
The fact I can compile/build realsense 2.0.0 with librealsense 2.x.x is not so important, sry for this.

jlgarcia75, 2017-09-29:
Hi @KappesR,

Can you provide us the details of the computer you are using with these cameras? Note that the ZR300 has been validated only on the Joule 570x board with limited validation on the Up Board.

To verify that the R200 camera works, try connecting the R200 to the USB 3.0 port of a Windows 10 PC and check if it works with the Windows SDK sample apps, found [here](https://registrationcenter.intel.com/en/forms/?productid=2809). Wait a few minutes after connecting the camera so the driver can install. If the driver does not install automatically, you can download it [here](https://downloadmirror.intel.com/25044/eng/intel_rs_dcm_r200_2.1.27.2853.exe).

Regards,
Jesus
Intel Customer Support
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

KappesR, 2017-10-04:
Hi @jlgarcia75,

I'm use a Latitude 5480 (i5 7th gen.) with the above Ubuntu and kernel-version for all cameras.

Want you just to check it is a hardware issue of the R200 I have?
I ask because in Ubuntu, the R200 is running with librealsense version 1.12.1, but in dmesg the uvcvideo driver show these messages everytime when I call `roslaunch realsense_camera r200_nodelet_default.launch`:
```
[  465.656755] uvcvideo: Non-zero status (-71) in video completion handler.
[  465.664756] uvcvideo: Non-zero status (-71) in video completion handler.
[  465.664819] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
[  465.715523] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 7: -32 (exp. 2).
```
That's my first problem because the driver has some problems.

My second problem is that `roslaunch realsense_camera r200_nodelet_rgbd.launch` crashes instantly and the parameter `enable_ir:=true` or `enable_pointcloud:=true` does not effect the other launch script `roslaunch realsense_camera r200_nodelet_default.launch`.
I got the depth and rgb stream as well as the camera_infos for both, but I can't get any information about the ir stream or pointcloud in this manner.

-------------------------------------------------------------
I tried it with the Windows SDK as you described.
The examples worked for the R200 on a NUC. But if a example failed for some reason, any further execution of the failed example or other examples failed as well.
I needed to replug the R200 and it worked again.

jlgarcia75, 2017-10-04:
@KappesR,

Did you try to patch the uvcvideo driver with /librealense/scripts/patch-uvcvideo-ubuntu-mainline.sh? After applying this patch, type: sudo modprobe uvcvideo. 

If this doesn't work, your camera or connection may have a hardware problem. Have you tried connecting the camera via a powered USB 3.0 HUB?

The R200 camera is discontinued so if it is a hardware problem and you are still within your warranty period, you can get a refund from your point of purchase.

Jesus G.
Intel Customer Support

KappesR, 2017-10-05:
@jlgarcia75, 

yes but I tested it again on a fresh Ubuntu 16.04 (same as everytime), but it did not work at all.
R200 is not connected.

I followed the [installation guide](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md) for tag v1.12.1 and it pointed out the script you mentioned works for Ubuntu 14.04:
> 
>     Update to 4.4 Stable Kernel
>         Run the following script to install necessary dependencies (GCC 4.9 compiler and openssl) and update kernel to v4.4-wily
>
>             ./scripts/install_dependencies-4.4.sh
>
>         Run the following script to patch uvcvideo.ko
>
>             ./scripts/patch-uvcvideo-4.4.sh v4.4-wily (note the argument provided to this version of the script)
>
>             This script involves shallow cloning the Linux source repository (~100mb), and may take a while
>     (OR) Keep 3.19.xx Kernel
>         Run the following script to patch uvcvideo.ko
>
>             ./scripts/patch-uvcvideo-ubuntu-mainline.sh
>
>         (R200 Only with 3.19.xx Kernel) Install connectivity workaround. Note this script needs to be run from within the scripts folder:
>
>
>             cd scripts
>             ./install-r200-udev-fix.sh
>             cd ..
>
>             This udev fix is not necessary for kernels >= 4.2
>             Use of 3.19.xx Kernel is not recommended.
Nethertheless, I run the script and I got the message: `Error :: You must put some 'source' URIs in your sources.list`
So, I followed this:
> (OR) Ubuntu 14.04.5 or Ubuntu 16.04.xx (Kernel 4.4)
> 
>     ./scripts/patch-uvcvideo-16.04.simple.sh
I started the script and after setting the configurations it terminated with the error:
```
scripts/sign-file.c:23:30: fatal error: openssl/opensslv.h: file or directory not found
compilation terminated.
scripts/Makefile.host:91: recipe for target „scripts/sign-file“ failed
make[2]: *** [scripts/sign-file] error 1
Makefile:566: recipe for target „scripts“ failed
make[1]: *** [scripts] error 2
Makefile:531: recipe for target „__build_one_by_one“ failed
make: *** [__build_one_by_one] error 2
```
I remembered the last time I left this error message as it is, because it worked nethertheless.
This time, I run the following to fix this:
`sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev libjpeg8-dev zlib1g-dev`
I run the `./scripts/patch-uvcvideo-16.04.simple.sh` again and the previous error is gone, but a new one appeared:
```
  MODPOST 1 modules
WARNING: "media_entity_cleanup" [/home/[user]/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_init" [/home/[user]/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_create_link" [/home/[user]/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
  CC      /home/[user]/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.mod.o
  LD [M]  /home/[user]/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
make: Directory „/home/[user]/librealsense/ubuntu-xenial“ leave
Unloading existing uvcvideo driver...
modprobe: ERROR: could not insert 'uvcvideo': Exec format error
```
dmesg output is (while the ):
```
[ 1251.988300] usbcore: deregistering interface driver uvcvideo
[ 1252.332956] media: Linux media interface: v0.10
[ 1252.339834] Linux video capture interface: v2.00
[ 1252.347285] uvcvideo: version magic '4.4.83+ SMP mod_unload ' should be '4.10.0-35-generic SMP mod_unload '
```
Is it just a problem with the kernel-version (`uname -r: 4.10.0-35-generic`)?

Thanh-Binh, 2017-10-05:
I have no success in Ubuntu with any RS camera...I also have the following errors:
ERROR [140028983531328] (types.h:123) uvcvideo kernel module is not loaded Last Error: Success
RealSense error calling rs2_query_devices(context:0x202cd70):
    uvcvideo kernel module is not loaded Last Error: Success

ev-mp, 2017-10-08:
@KappesR ,
You need to follow the installation instructions from the [master branch](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) for installing the backend and patching the kernel.
The instructions you refer to apply for the 4.4 kernel that was current when the tag was issued (Nov 2016)
> I followed the installation guide for tag v1.12.1 and it pointed out the script you mentioned works for Ubuntu 14.04:

Please reinstall the backend accordingly and update

KappesR, 2017-10-11:
@ev-mp ,
thanks also!

I tried as you mentioned on a clean `Ubuntu 16.04` with `4.10.0-35-generic`.
I opened qv4l2 which is a tool for video grapping and can be used as well as the ros realsense nodes.
So, the dmesg output with this tool is as follow: 
```
[  165.989929] usb 2-3: new SuperSpeed USB device number 6 using xhci_hcd
[  166.012596] usb 2-3: New USB device found, idVendor=8086, idProduct=0a80
[  166.012599] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  166.012602] usb 2-3: Product: Intel RealSense 3D Camera R200
[  166.012604] usb 2-3: Manufacturer: Intel Corp
[  166.012606] usb 2-3: SerialNumber: SN_2521007700
[  166.016238] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  166.020329] uvcvideo 2-3:1.0: Entity type for entity Extension 2 was not initialized!
[  166.020336] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[  166.021300] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  166.025502] uvcvideo: Unable to create debugfs 2-6 directory.
[  166.025595] uvcvideo 2-3:1.2: Entity type for entity Camera 4 was not initialized!
[  166.025875] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.2/input/input16
[  166.026961] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  166.031024] uvcvideo: Unable to create debugfs 2-6 directory.
[  166.031115] uvcvideo 2-3:1.4: Entity type for entity Processing 7 was not initialized!
[  166.031119] uvcvideo 2-3:1.4: Entity type for entity Camera 6 was not initialized!
[  166.031355] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.4/input/input17
[  204.455146] uvcvideo: Failed to query (GET_CUR) UVC control 6 on unit 7: -32 (exp. 2).
[  204.456210] uvcvideo: Failed to query (GET_CUR) UVC control 4 on unit 6: -32 (exp. 4).
```
Still, the same problem.
Without the use of the ros realsense nodes the error appears also.

The view:
[image]
I got two errors from this programm:
```
Error Hue: Input/Output error
Error Exposure (Absolute): Input/Output error
```
In which way this correlate to the dmesg output I can only guess.
Maybe there is still a problem with the uvcvideo update or a problem with the firmware?

@jlgarcia75,
I tried to use the warranty three weeks ago, but it was denied.
I hope it's ok to publish this here:
[image]

The reasoning was as follow:
```
You can check the status of your Return by logging into your account.
This RMA has been denied. The product is sold AS-IS as stated on the website. There is no warranty support.
```
I already ordered a USB 3.0 Hub with powered ports, but it's curious that two R200 broke in one week.
And that the ZR300 has the same errors as well..

jlgarcia75, 2017-10-11:
@KappesR,

The return period for RealSense cameras is 90 days so if you purchased them over 90 days ago that is why your request is denied.  Or, I noticed that your request is for an "Exchange." R200 cameras are discontinued and there are no more in stock. If you purchased it within 90 days then you can get a refund, but not an exchange.

Have you tried the USB HUB yet? It may solve your problems. It is odd for all three cameras to fail simultaneously unless it is a USB issue.

Jesus
Intel Customer Support

KappesR, 2017-10-11:
@jlgarcia75, 

I see, we bought them one year and two months ago, so that is the reason for the denied request.

I tried the USB Hub now but same error log, any further ideas?
Thanks for the help so far!

ev-mp, 2017-10-16:
@KappesR , please perform  several troubleshooting and data gathering steps:
1. Use the device with Windows OS or some other PC and check whether the same issues can be reproduced
2. Run `dmesg -wT` and then do the following:
  - Connect R200 to the USB port.
  - Run `cpp-tutorial1-depth` demo in a separate terminal
  - Check whether the device is streaming and also collect and share output from both terminals

KappesR, 2017-10-16:
@ev-mp,

thanks for the steps:

**1.**
I have done this before:
> I tried it with the Windows SDK as you described.
> The examples worked for the R200 on a NUC. But if a example failed for some reason, any further execution of the failed example or other examples failed as well.
> I needed to replug the R200 and it worked again.

Its not the same behavior.

**2.**
I have tested cpp-tutorial-1-depth and cpp-tutorial-2-streams.

cpp-tutorial-1-depth:
Console Output:
```
                                                                
                                                                
                                                                
                                    .n.::.                      
                              .       :hWWXh.                   
                              .:.      X. hWB.                  
                              hh       :  nWh:                  
                            .:.        .  BWhh                  
                            :X          ::WBXB                  
                                             n                  
                             .              :n                  
                             .             h:                   
                            .n             h                    
                             :   .         h                    
                                           :: ...               
                     .nXBhBn:BBBnnn                   .nhh..    
                  :nWWWWWWBBXnhBhn:                     .   h   
              .:BWXWB:  ..n:                                h   
         :. ::.   ..     :n                                 h   
       :h.      .:                                              
                .                      .                        

                                                                
                                                                
                                                                

```

While dmesg -wT:
```
[Mo Okt 16 15:25:13 2017] usb 2-3: new SuperSpeed USB device number 10 using xhci_hcd
[Mo Okt 16 15:25:13 2017] usb 2-3: New USB device found, idVendor=8086, idProduct=0a80
[Mo Okt 16 15:25:13 2017] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mo Okt 16 15:25:13 2017] usb 2-3: Product: Intel RealSense 3D Camera R200
[Mo Okt 16 15:25:13 2017] usb 2-3: Manufacturer: Intel Corp
[Mo Okt 16 15:25:13 2017] usb 2-3: SerialNumber: SN_2521001926
[Mo Okt 16 15:25:13 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Mo Okt 16 15:25:13 2017] uvcvideo 2-3:1.0: Entity type for entity Extension 2 was not initialized!
[Mo Okt 16 15:25:13 2017] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[Mo Okt 16 15:25:13 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Mo Okt 16 15:25:13 2017] uvcvideo: Unable to create debugfs 2-10 directory.
[Mo Okt 16 15:25:13 2017] uvcvideo 2-3:1.2: Entity type for entity Camera 4 was not initialized!
[Mo Okt 16 15:25:13 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.2/input/input24
[Mo Okt 16 15:25:13 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Mo Okt 16 15:25:13 2017] uvcvideo: Unable to create debugfs 2-10 directory.
[Mo Okt 16 15:25:13 2017] uvcvideo 2-3:1.4: Entity type for entity Processing 7 was not initialized!
[Mo Okt 16 15:25:13 2017] uvcvideo 2-3:1.4: Entity type for entity Camera 6 was not initialized!
[Mo Okt 16 15:25:13 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.4/input/input25
```
In several runs, no error appeared.

cpp-tutorial-2-depth:
[... 44 lines cut ...]
The second run:
```
[Mo Okt 16 15:33:28 2017] usb 2-3: USB disconnect, device number 10
[Mo Okt 16 15:33:31 2017] usb 2-3: new SuperSpeed USB device number 11 using xhci_hcd
[Mo Okt 16 15:33:31 2017] usb 2-3: New USB device found, idVendor=8086, idProduct=0a80
[Mo Okt 16 15:33:31 2017] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mo Okt 16 15:33:31 2017] usb 2-3: Product: Intel RealSense 3D Camera R200
[Mo Okt 16 15:33:31 2017] usb 2-3: Manufacturer: Intel Corp
[Mo Okt 16 15:33:31 2017] usb 2-3: SerialNumber: SN_2521001926
[Mo Okt 16 15:33:31 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Mo Okt 16 15:33:31 2017] uvcvideo 2-3:1.0: Entity type for entity Extension 2 was not initialized!
[Mo Okt 16 15:33:31 2017] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[Mo Okt 16 15:33:31 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Mo Okt 16 15:33:31 2017] uvcvideo: Unable to create debugfs 2-11 directory.
[Mo Okt 16 15:33:31 2017] uvcvideo 2-3:1.2: Entity type for entity Camera 4 was not initialized!
[Mo Okt 16 15:33:31 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.2/input/input26
[Mo Okt 16 15:33:31 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Mo Okt 16 15:33:31 2017] uvcvideo: Unable to create debugfs 2-11 directory.
[Mo Okt 16 15:33:31 2017] uvcvideo 2-3:1.4: Entity type for entity Processing 7 was not initialized!
[Mo Okt 16 15:33:31 2017] uvcvideo 2-3:1.4: Entity type for entity Camera 6 was not initialized!
[Mo Okt 16 15:33:31 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.4/input/input27
[Mo Okt 16 15:33:42 2017] uvcvideo: Non-zero status (-71) in video completion handler.
```

On the second start of `cpp-tutorial-2-streams` the log ` [Mo Okt 16 15:33:47 2017] uvcvideo: Non-zero status (-71) in video completion handler.` appears only once or three times. After this, it appears everytime two times by restarting the binary.

ev-mp, 2017-10-16:
Hi @KappesR, according to the logs and the provided screenshots:
1. Device enumeration is performed correctly (USB3 device).
2. Kernel module is properly configured (no missing formats).
3. Librealsense streams and renders all R200-available sensors.
The message :

> uvcvideo: Non-zero status (-71)

 is a warning and can be safely disregarded.

I cannot immediately comment on ros-node log  as you should probably refer it to https://github.com/intel-ros/realsense , but there are hints for the package configuration issues:

> ResourceNotFound: rgbd_launch

But other than that the software stack is properly configured and fully functional, and ,actually, I'd recommend to use this specific host as a reference system to validate the two additional R200 that you've mentioned in your previous posts.

As to the scenario that the device stops streaming and requires a manual re-plug - it is definitely an issue on a Host-Device level,  and without having kernel log and a deterministic reproduction flow it would be a pure speculative to determine what went wrong.
Based on previous experience the R200 family sensible to cables length/quality and the power. In addition the host USB controllers from different vendor matters.

Please run a comprehensive search on the issues archive to obtain additional community inputs and reccomendations, e.g [(#117)](https://github.com/IntelRealSense/librealsense/issues/117), [(#67)](https://github.com/IntelRealSense/librealsense/issues/67), [(#418)](https://github.com/IntelRealSense/librealsense/issues/418) and more

KappesR, 2017-10-17:
@ev-mp,

i tried further test programs of the librealsense.
I can reproduce the `uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1)` with the following library example:

cpp-config-ui:

Console:
```
~$ cpp-config-ui:
Capturing DEPTH at 628 x 468, fov = 56.6 x 43.7, distortion = NONE
Capturing COLOR at 640 x 480, fov = 55.0 x 42.3, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 640 x 480, fov = 57.5 x 44.7, distortion = NONE
Capturing INFRARED2 at 640 x 480, fov = 57.5 x 44.7, distortion = NONE
```

Dmesg -wT:
```
[Di Okt 17 13:50:52 2017] usb 2-3: new SuperSpeed USB device number 8 using xhci_hcd
[Di Okt 17 13:50:52 2017] usb 2-3: New USB device found, idVendor=8086, idProduct=0a80
[Di Okt 17 13:50:52 2017] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Di Okt 17 13:50:52 2017] usb 2-3: Product: Intel RealSense 3D Camera R200
[Di Okt 17 13:50:52 2017] usb 2-3: Manufacturer: Intel Corp
[Di Okt 17 13:50:52 2017] usb 2-3: SerialNumber: SN_2521001926
[Di Okt 17 13:50:52 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Di Okt 17 13:50:52 2017] uvcvideo 2-3:1.0: Entity type for entity Extension 2 was not initialized!
[Di Okt 17 13:50:52 2017] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[Di Okt 17 13:50:52 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Di Okt 17 13:50:52 2017] uvcvideo: Unable to create debugfs 2-8 directory.
[Di Okt 17 13:50:52 2017] uvcvideo 2-3:1.2: Entity type for entity Camera 4 was not initialized!
[Di Okt 17 13:50:52 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.2/input/input20
[Di Okt 17 13:50:52 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Di Okt 17 13:50:52 2017] uvcvideo: Unable to create debugfs 2-8 directory.
[Di Okt 17 13:50:52 2017] uvcvideo 2-3:1.4: Entity type for entity Processing 7 was not initialized!
[Di Okt 17 13:50:52 2017] uvcvideo 2-3:1.4: Entity type for entity Camera 6 was not initialized!
[Di Okt 17 13:50:52 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.4/input/input21
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:07 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:08 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
[Di Okt 17 13:51:08 2017] uvcvideo: Failed to query (GET_CUR) UVC control 18 on unit 2: -32 (exp. 1).
```
Screen:
[image]

The errors appear after i press **start capture**, for some time the capturing does not start. After the uvcvideo driver stop failing, the video capture is working.

I will look into the closed issues you mentioned and let you know if i could fix it with them, thereby I was able to reproduce the issue with the librealsense examples, with qv4l2 and with the ROS-Wrapper it is a independent issue to the ROS Wrapper and to the librealsense library. Maybe the uvcvideo driver update is not working correctly.

jlgarcia75, 2017-10-26:
Hello @KappesR,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

KappesR, 2017-10-30:
Hi @jlgarcia75,

yes, I tried to solve the problem with the attached issues by @ev-mp and it didn't worked.
I search further in the issue list of the ros-wrapper but couldn't solved this up yet.

If you have some further suggestions, I would be glad.

ev-mp, 2017-11-01:
Hi @KappesR , I'm not sure that I'm following the issue with you :  Initially you indicated that all three devices stopped working,  but then you did provide streaming captures for at least some of the devices. 
Can you clarify what are the currently unresolved issues are ? thanks

HongweiXie, 2017-11-15:
I have met the same problem on Ubuntu 16.04.3? do you have a solution now?

fxprime, 2020-08-02:
Me too, crash when set parameter
error that occur is 

```
RealSense error calling rs_device_supports_option( .... 
```