realsenseai/librealsense #667: Zr300 supported system.

Hutch118, 2017-09-29:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/supp
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | ?  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |   |

With the Intel Joule now discontinued what are the "supported" systems to use with the zr300?

jlgarcia75, 2017-10-02:
Hello @Hutch118,

For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

The ZR300 has limited validation on the UP Board (https://up-shop.org/up-boards/62-up-board-4b-32-gb-emmc-memory.html) with active heat sync. Note that the Intel® RealSense™ ZR300 has been discontinued and customers are encouraged to move to the Intel® RealSense™ Depth Camera D400 series when they are released. 

Best regards,
Jesus
Intel Customer Support

Hutch118, 2017-10-02:
I appreciate it I had no idea the zr300 was discontinued - along with the Joule. I bought both and had trouble with the Joule.  If I went for a D400 would it be discontinued within the year? I'm not understanding the product lifecycles. Feels like I can't dev quick enough to use Intel products.

jlgarcia75, 2017-10-02:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

I cannot comment on how long Intel products will remain on the market but I can tell you that there are currently no plans to discontinue the Intel® RealSense™ Depth Camera D400 series within one year.

Best regards,
Jesus
Intel Customer Support

Hutch118, 2017-10-02:
Thanks for your time.

cvilas, 2017-10-25:
Oh dear! ZR300 is already discontinued? How can we rely on Intel devices to develop a product if they are not going to be supported even for a year?

Hutch118, 2017-10-25:
I can't really tell if the zr300 is gone or not.  No release for it that I see. Sites are not showing notice. It doesn't seem to be supported in the new librealsense but the sr300 is..? The zr300 is in the euclid and still for sale..?  Confusing

alfredanil, 2017-10-25:
Euclid is back in stock, what is going on? if it is not going to be suppported in future why is intel still restocking it? Clearly there are issues with the hardware especially with the heat buildup and throttling. I would like to know if Movidius™ Neural Compute Stick can improve the performance of Euclid? if so, how?

jlgarcia75, 2017-10-26:
Hi,

The ZR300 is [discontinued ](http://qdms.intel.com/dm/i.aspx/56127FAC-E173-46EE-AA01-A895A0A678CA/PCN115512-00.pdf) but will continue to be sold until inventory runs out. The software for it is no longer supported. Euclid is back in stock until inventory runs out. Both of these products have limited support since their software is no longer being developed or fixed.

We are transitioning to the RealSense Depth Cameras D415 and D435.

Regards,
Jesus
Intel Customer Support

closed 2017-10-26 by jlgarcia75

alfredanil, 2017-10-27:
What do you expect from the developers and investors who invested in ZR300 and Euclid? Do we get a refund or some kind of compensation for transitioning to D series?

Hutch118, 2017-10-27:
Alfredanil I had two development products get pulled ZR300 and Joule. I haven't even seen even a discount for dev's that bought a product that was discontinued within a year. I see no point in trying the D series no conciliation for short lived products no gaurentee on support life. I'm moving on to other companies.

alfredanil, 2018-02-22:
@AngelCrusher 
That's true although D415 looks promising. I guess it's wait and watch for now.