realsenseai/librealsense #684: Depth deviation over time

shapilov3000, 2017-10-04:
We are using SR300 and trying to estimate precision of the cameras.
I fixed one SR300 in front of a wall and was writing depth of several points to file during 2+ hours.
The result (horizontal axis: time in minutes, vertical axis: depth*scale in meters):
[image]

So my question is: what causes such deviation of depth values, is it heating of the camera or some other physical process? Is it possibe to correct that error? If heating is the reason, how to get internal camera temperature?

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0 | 
| Operating System & Version | Windows 10 x64 |
| Kernel Version (Linux Only)  |   |

shapilov3000, 2017-10-05:
Nobody knows?
10 millimeters error in 30 minutes for ~500 millimeters to object it's about 2%.
Although according the documentation error is less than 1%.

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).
@shapilov3000,

This is very odd behavior. 

Which model of processor does your PC have? 

How are you measuring depth?

Best regards,
Jesus
Intel Customer Support

shapilov3000, 2017-10-09:
My processor is Intel(R) Core(TM) i5-7500 CPU @ 3.40GHz
All tested cameras' firmware version is 3.21.0.0
I took depth values from the depth stream.
As I understand, the depth deviation means that the corresponding point is moving while camera is working (even if the camera and the object are static).

More graphs (3 more cameras):
[image]

As we can see, there are some levels of the depth values and the point moves between them:
[image]

Do you know if it is possible to correct that deviation?

jlgarcia75, 2017-10-09:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).
@shapilov3000,

You are right, according to my research, this fluctuation in depth readings may be temperature related. The discussion in this post may be helpful: https://github.com/IntelRealSense/librealsense/issues/96.
 
Best regards,
Jesus
Intel Customer Support

shapilov3000, 2017-10-09:
@jlgarcia75 ,
thank you for your response and for the link

We were trying to scan static objects with SR300 and expecting to reach at least 1 millimeter occurancy.
How do you think, would we get stablier result if we cool sensors directly?

Another decision is turning on cameras for short time period and turning them off for cooling. I tested this idea with 10 sec of working and 140 sec of resting periods for about 3 hours and got this result (horizontal axis: frames, vertical axis: depth in meters):
[image]

Well... it's more or less stable. May be this would be usefull for somebody.
So what do you think about direct cooling?

jlgarcia75, 2017-10-09:
@shapilov3000,

The results look much better with your protocol. Howeever, I cannot predict what kind of results you will get with direct cooling. This has not been tested. The camera is supposed to be stable if operated within the acceptable temperature ranges (see page 30 of the [RealSense SR300 DataSheet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/realsense-sr300-datasheet1-0.pdf))

What kind of objects are you scanning? Size? Material?

shapilov3000, 2017-10-10:
@jlgarcia75,
We are scanning plastic objects, cloth objects and parts of human body, our objects are 20-30 santimeters, scanning distance is 30-60 santimeters.
I know it's hard to reach 1 millimeter accuracy with SR300 in our cases, would it be possible with new 400 cameras?

Jiloc, 2017-10-10:
@jlgarcia75 similar use case here. We too need a 1 millimeter accuracy and would like to know if the new D400 are able to reach it.

@shapilov3000 could I ask you what did you integrate with librealsense to achieve 3D scanning? We need to move from the old Windows SDK and we are looking for something reliable

shapilov3000, 2017-10-10:
@Jiloc,
we are using our own soft with librealsense

Jiloc, 2017-10-10:
@shapilov3000 I mean, did you integrate a 3D scanning library as [PCL](https://github.com/PointCloudLibrary/pcl) / [InfiniTAM](https://github.com/victorprad/InfiniTAM) or did you write your own implementation of a 3D scanning algorithm? If the latter, did you implemented KinFu or did you find a better solution? 

Of course feel free to say that it's private info if you can't answer!

shapilov3000, 2017-10-10:
@Jiloc yes, we built a specific scanning device with a specific architecture and implemented our scanning and processing algorithms. Sorry, I can't say a lot, we don't sell our soft or devices.
Fundamentally, we invoke librealsense methods from our code.

shapilov3000, 2017-10-16:
@jlgarcia75,
do you know how accurate D400 are?

Best regards,
Oleg Shapilov

jlgarcia75, 2017-10-17:
@shapilov3000,

I'm trying to get this information for you. It is not published. Stay tuned.

Jesus
Intel Customer Support

jlgarcia75, 2017-10-19:
Hello,

The accuracy of the D400 series cameras varies slightly with distance but the error is <1% of distance. For example, if the object is 500mm away, you can expect accuracy <5mm. Of course this all depends on environmental factors such as lighting, material, etc. You can use the [Depth Quality Tool](https://github.com/IntelRealSense/librealsense/tree/development/tools/depth-quality) to measure depth accuracy.

I hope this helps.

Regards,
Jesus G.
Intel Customer Support

shapilov3000, 2017-10-20:
@jlgarcia75, thank you for your response and usefull infomation.

Could you tell, please, if depth measurement of the D400 is stable and equal indoors and outdoors with bright sun?
And if so, how do D400 work? Do the cameras project some IR pattern (as SR300) or use object texture?
Do several cameras interfere?
We need to know if the new cameras are more usefull for us than SR300.

Best regards,
Oleg Shapilov

jlgarcia75, 2017-10-20:
Hello @shapilov3000,

Yes, it is stable indoors and outdoors, with longer range outdoors.  D400 works with both IR and Visible light. For Indoors, it has the IR projector to illuminate scenes with low texture and if IR is not reflected on certain surfaces such as carpets, the stereo camera works off of visible light. The projected pattern is randomized dots and so they do not interfere with multiple cameras, unlike TOF/Structured Light based depth cameras. In fact, it works better as overlaps creates more dots/texture.

Regards,
Jesus G.
Intel Customer Support

closed 2017-10-26 by jlgarcia75