realsenseai/librealsense #685: SR300 auto disconnected while i open the IR camera on the viewer,

axistudio, 2017-10-04:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |

dorodnic, 2017-10-05:
Hi @axistudio 
This is most certainly not the expected behavior. However, I suspect the camera is drawing more power then what the USB port can provide. Can you test this on another PC with USB3 or on a different USB port?

Jiloc, 2017-10-05:
Same behaviour here with 2 different computers. With the third one it works flawlessly.

axistudio, 2017-10-05:
i did fix it by using a powerd usb hub , it should be the power draw from sensor is high.

closed 2017-10-06 by jlgarcia75