realsenseai/librealsense #686: Anychance we can have unity SDK later on this version?

axistudio, 2017-10-04:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |   Anychance we can have unity SDK later on this version? |
|---|---|
| Camera Model | SR300| 
| Firmware Version |   | 
| Operating System & Version | win 10  |
| Kernel Version (Linux Only)  |   |

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello @axistudio,

The RealSense SDK for Windows is no longer being developed or supported. However, support for the SR300 and D400 series cameras will continue in the Intel RealSense SDK 2.0 for Linux. This SDK is still in development phase and the engineering team is still working on features. We cannot comment on whether a Unity wrapper will be written for it by Intel but I will certainly submit your idea for consideration. 

Best regards,
Jesus
Intel Customer Support

jlgarcia75, 2017-10-12:
@axistudio Can you provide us more info regarding your requirements and usages for Unity and the SR300 camera?

closed 2017-10-17 by jlgarcia75

mflux, 2017-12-18:
I'm also here for Unity + RS (D400 camera to be precise).

I want to access the IR camera stream directly and use that to look at IR trackers for augmented reality. The old SDK is still floating around on the internet, will that be compatible with any of the existing line of Realsense cameras?