realsenseai/librealsense #701: Python wrapper 'config' is ignored

dougsouza, 2017-10-12:
Hello,

I am using the python wrapper to manipulate data with opencv. The lastest [example](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/opencv_viewer_example.py) seems cool but it doesn't work in practice. For the code below:

```python
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline_profile = pipeline.start(config)
```
if I run:
```python
print pip_profile.get_streams()
```
I get:
```python
[<pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 60fps Z16>, <pyrealsense2.video_stream_profile: Color(0) 1920x1080 @ 30fps Y16>]
```
Meaning the config I just passed is completely ignored, and indeed, when I capture the the RGB and Depth data they do not follow the settings I passed on the config object. Any ideia how to solve this?

Thanks!

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04 LTS   |
| Kernel Version (Linux Only)  |   |

zivsha, 2017-10-14:
Hi @dougsouza,
Thanks for reporting this behavior, this is a bug at the C++ level.
I will update when a PR is merged

zivsha, 2017-10-15:
@dougsouza , the PR is now merged to development. 
Please let us know if you have any additional issues.

dougsouza, 2017-10-16:
@zivsha, thanks for the support!! I am going to rebuild the library and try again.

Will reopen if needed.

dougsouza, 2017-10-24:
@zivsha, in your commit 6cf290bc8a40bccc44c85c280f2565dab7338831 you forgot to pass the image format as well. Therefore in this overload of the method the format is still being ignored.

Cheers,

Douglas

zivsha, 2017-10-24:
@dougsouza , you are correct. PR #734 fixed that one.

dougsouza, 2017-10-31:
@zivsha, yes. Sorry I missed that.

closed 2017-10-31 by dougsouza