realsenseai/librealsense #704: 3D model generation

wolffadam, 2017-10-15:
I have a SR300 camera and I tried to use the new 2.0 SDK. with the Viewer I noticed I can generate a 3D model, but the only format I can save it is Polygon File Format, and I don't know how to deal with this kind of format.
I tried to convert it to STL file but it didn't work.
Is there a way to save the model in different format?

zivsha, 2017-10-16:
Hi @wolffadam ,

The web provides many viewing tools for such files. A quick search landed me on [meshlab](http://www.meshlab.net/) which supports this format. You are free to try any tool you wish to open it, and I suppose there are tools for converting PLY to other known formats.

At the moment the only format that the viewer supports for saving 3D models is to the [PLY file format](https://en.wikipedia.org/wiki/PLY_(file_format)). This is a well known format for 3D meshes.

closed 2017-10-26 by jlgarcia75

hdjsjyl, 2018-03-06:
@zivsha Thanks for SDK. And I have a problem. I am using D435, After I save 3D mdoel to .ply file, I use meshlab to open .ply, but I just can see outline of tripod and can not see the people outline. Any advice will be appreciated, I am looking forward to your reply. Thank you very much.
[image], 
[image]

baptiste-mnh, 2018-05-03:
Try to crop the PLY to only keep this part : 
[image]
I'm pretty sure you're there.