realsenseai/librealsense #707: CPU utilization 95+%

JokerShao, 2017-10-16:
**Note:** Please take advantage of RealSense [Community](https://communities.intel.com/community/tech/realsense) and [Support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html) sites.

| Required Info  |  

When i read a aligned stream like this:
uchar* pCad = (uchar*)dev.get_frame_data(rs::stream::color_aligned_to_depth);

the CPU utilization 95+%.

 |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.10  | 
| Operating System & Version |  Windows10  |
| Kernel Version (Linux Only)  |   |

jlgarcia75, 2017-10-17:
Hello @JokerShao,

Please provide more system details. Which processor model do you have? Do you get this high utilization with the librealsense samples? What else is running when this happens?

Regards,
Jesus
Intel Customer Support

JokerShao, 2017-10-19:
Thanks for your reply,

I have Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz  3.40GHz.
The same problem also appear when i run the librealsense samples.
Seems like I found the reason.

In image.cpp line 522
```#pragma omp parallel for schedule(dynamic)```
When I comment out this code, the utilization drop to 25+%.
But the speed running these code seems not slow down.
Whether comment or not, the time run these code is equal, about 15ms.
**WHY?**
```
#pragma omp parallel for schedule(dynamic)
        for(int depth_y = 0; depth_y < depth_intrin.height; ++depth_y)
        {
            int depth_pixel_index = depth_y * depth_intrin.width;
            for(int depth_x = 0; depth_x < depth_intrin.width; ++depth_x, ++depth_pixel_index)
            {
                // Skip over depth pixels with the value of zero, we have no depth data so we will not write anything into our aligned images
                if(float depth = get_depth(depth_pixel_index))
                {
                    // Map the top-left corner of the depth pixel onto the other image
                    float depth_pixel[2] = {depth_x-0.5f, depth_y-0.5f}, depth_point[3], other_point[3], other_pixel[2];
                    rs_deproject_pixel_to_point(depth_point, &depth_intrin, depth_pixel, depth);
                    rs_transform_point_to_point(other_point, &depth_to_other, depth_point);
                    rs_project_point_to_pixel(other_pixel, &other_intrin, other_point);
                    const int other_x0 = static_cast<int>(other_pixel[0] + 0.5f);
                    const int other_y0 = static_cast<int>(other_pixel[1] + 0.5f);

                    // Map the bottom-right corner of the depth pixel onto the other image
                    depth_pixel[0] = depth_x+0.5f; depth_pixel[1] = depth_y+0.5f;
                    rs_deproject_pixel_to_point(depth_point, &depth_intrin, depth_pixel, depth);
                    rs_transform_point_to_point(other_point, &depth_to_other, depth_point);
                    rs_project_point_to_pixel(other_pixel, &other_intrin, other_point);
                    const int other_x1 = static_cast<int>(other_pixel[0] + 0.5f);
                    const int other_y1 = static_cast<int>(other_pixel[1] + 0.5f);

                    if(other_x0 < 0 || other_y0 < 0 || other_x1 >= other_intrin.width || other_y1 >= other_intrin.height) continue;

                    // Transfer between the depth pixels and the pixels inside the rectangle on the other image
                    for(int y=other_y0; y<=other_y1; ++y) for(int x=other_x0; x<=other_x1; ++x) transfer_pixel(depth_pixel_index, y * other_intrin.width + x);
                }
            }
```

Thanks a lot,
Best wishes for you.

jlgarcia75, 2017-10-19:
Hello @JokerShao,

I found this page which describes what that #pragma line does. https://software.intel.com/en-us/articles/openmp-loop-scheduling

dynamic | Use the internal work queue to give a chunk-sized block of loop iterations to each thread. When a thread is finished, it retrieves the next block of loop iterations from the top of the work queue. By default, the chunk size is 1. **Be careful when using this scheduling type because of the extra overhead involved.**

It seems that there is extra overhead when (dynamic) is used.

Because of the large amounts of data required to the depth processing, RealSense applications tend to have a high CPU utilization. Even I see that on my 5th Gen Intel Core i5 system. The R200 cameras have data coming from two depth sensors and one RGB sensor. That is much more than what you get from a regular RGB camera. 

Regards,
Jesus G.
Intel Customer Support

skohlbr, 2017-10-25:
Using `#pragma omp parallel for schedule(dynamic)` can have very significant (negative) performance impact. In another application (here: https://github.com/ros-planning/moveit_ros/issues/563) it caused a 320% CPU load, compared to ~20% when not used.

dorodnic, 2017-10-27:
Related to #744
Please note that you can choose to disable **OpenMP** using [CMake build configuration](https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration).
@skohlbr, thanks for your comment. What mode of optimization would you recommend?

closed 2017-11-01 by jlgarcia75