realsenseai/librealsense #721: Accessing infrared stream

Jason420Git, 2017-10-19:
| Required Info  |   |
|---|---|
| Camera Model |  D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |

Hi,

I am following the API guide but having issue getting Infrared stream. The method: rs2::device::get() does not take any parameters which is unlike the example. Instead there are just get_depth and get_color.

zivsha, 2017-10-20:
Hi,

`rs2::device::get()` returns a `std::shared_ptr<rs2_device>` to the underlying device used inside the library. You can't use this pointer for anything in your application other than passing it to the `C` API, which we don't recommend mixing with the `C++` API.

Are you following this guide: ["API How To"](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers)?

Please try the suggested code in the [API How To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To) guide (found in our Wiki) and let us know if you managed to get what you need.

Jason420Git, 2017-10-21:
I am following this yes. But there is not an example of how to get a frame from the infrared stream. And the example which I presume i could follow, is no longer applicable. 

 rs2::frame depth_frame = frames.get(RS2_STREAM_INFRARED); 

Is what I presumed I could do (as per the example using depth) but it does no longer exist.
The method for getting depth frames is now frame.get_depth_frame(), but there is no call for infrared frames.

zivsha, 2017-10-21:
`rs2::frameset` provides several ways of getting a specific frame from it.

1.  Iterating over it a querying the type of stream per frame.
```cpp
rs2::frameset frames = pipe.wait_for_frames();
for(auto&& f : frames)
{
    if(f.get_profile().stream_type() == RS2_STREAM_INFRARED)
    {
        \\f is an infrared frame
    }
}
```

2. Using the `rs2::frameset::first(rs_stream s)` method:
https://github.com/IntelRealSense/librealsense/blob/67cdafef0c79554cd0c482d188f76feffd1bd9e2/include/librealsense2/hpp/rs_frame.hpp#L483
For example:
```cpp
rs2::frameset frames = pipe.wait_for_frames();
rs2::frame ir_frame = frames.first(RS2_STREAM_INFRARED);
```

Jason420Git, 2017-10-23:
Thank you. The .first is what I was missing.

zivsha, 2017-10-24:
FYI, the Wiki contains an example for this use case: [wait-for-coherent-set-of-frames](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#wait-for-coherent-set-of-frames)

closed 2017-10-26 by jlgarcia75