realsenseai/librealsense #722: Problem with glfw3

manhha1402, 2017-10-19:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 2   | 
| Operating System & Version |  14.04  |
| Kernel Version (Linux Only)  |  4.4.0-97-generic |
Hi,
I could not install  librealsense2-utils from instruction guide. It showed the error:
The following packages have unmet dependencies:
 librealsense2-utils : Depends: libglfw3 (>= 3.1) but 3.0.4-2~trusty is to be installed
E: Unable to correct problems, you have held broken packages.

I tried to remove glfw3 but the problem still happens
Any help?
Thanks

jlgarcia75, 2017-10-20:
Hello @manhha1402,

Follow the instructions here to install glfw3, https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md.

On Ubuntu 14.04 or when running of Ubuntu 16.04 live-disk, please use ./scripts/install_glfw3.sh

Regards,
Jesus G.
Intel Customer Support

manhha1402, 2017-10-20:
Hi,
Thanks. I installed successfully sdk2.0 from source(https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) .  Now I am able to capture data from SR300 using libv2 using tools from the sdk . But I could not find the library "#include <librealsense2/rsutils.hpp>" when trying to run the code "Getting Started with OpenCV". I run the code from the instruction guide :" sudo apt-get install librealsense2-utils" . It displayed the error:
 The following packages have unmet dependencies:
 librealsense2-utils : Depends: librealsense2 (= 2.8.0-0~realsense0.40) but it is not going to be installed
                       Depends: libglfw3 (>= 3.1) but it is not installable
Is there any problem here?
Can I install both librealsense v1 and v2 in a computer. Do they have any conflict ? 
Thanks

dorodnic, 2017-10-21:
Hi @manhha1402 
We will update the OpenCV instructions shortly. Please take a look at other C++ [tutorials](https://github.com/IntelRealSense/librealsense/tree/master/examples) until OpenCV example is back to normal. 
The package installer will be problematic to use on Ubuntu 14.04 since `glfw` dependency is only available as a package since Ubuntu 16.04. On Ubuntu 14.04 you can use the [installation instructions](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) to build the SDK from source.

closed 2017-10-26 by manhha1402

LZhipeng, 2017-12-03:
I have installed the glfw-3.2.1 on my ubuntu 14.04. But When I intall librealsense2-utils also the following error:
root@localhost:~# apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils : Depends: libglfw3 (>= 3.1) but 3.0.4-1 is to be installed
E: Unable to correct problems, you have held broken packages.
root@localhost:/usr/local/lib# ls -l
total 1456
drwxr-xr-x 3 root root    4096 Dec  1 00:34 cmake
-rw-r--r-- 1 root root  245806 Dec  1 00:34 libglfw3.a
lrwxrwxrwx 1 root root      12 Dec  1 15:26 libglfw.so -> libglfw.so.3
lrwxrwxrwx 1 root root      14 Dec  1 15:26 libglfw.so.3 -> libglfw.so.3.2
-rw-r--r-- 1 root root  163832 Dec  1 16:13 libglfw.so.3.2

How can resolve above dependency error?
Thanks!

DanielArnett, 2018-01-24:
One possible solution. I came here with the below error:

````
~/librealsense$ ./scripts/install_glfw3.sh 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version.
libglu1-mesa-dev is already the newest version.
libglu1-mesa-dev set to manually installed.
git is already the newest version.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 cmake : Depends: cmake-data (>= 2.8.12.2)
E: Unable to correct problems, you have held broken packages.
```

My solution was to open `./scripts/install_glfw3.sh` and comment out the first two lines such that it now reads:

```
#!/bin/bash -e
# Installs glfw3 from source as a shared lib

# Make sure glfw (or any previous version) is not already installed
#sudo apt-get update

#sudo apt-get install build-essential cmake git xorg-dev libglu1-mesa-dev

git clone https://github.com/glfw/glfw.git /tmp/glfw

cd /tmp/glfw

git checkout latest

cmake . -DBUILD_SHARED_LIBS=ON

make

sudo make install

sudo ldconfig

rm -rf /tmp/glfw

echo "Done installing glfw3!"
```

wucm1994, 2018-07-19:
Hi! I also encountered this problem and I have solved it in this way:

1. Download this install script for trusty: [libfreenect2](https://github.com/OpenKinect/libfreenect2/blob/master/depends/download_debs_trusty.sh)
2. Enter to the ./debs folder, use `sudo dpkg -i libglfw3*deb`
3. ` sudo apt-get install -f`

Then the unmet dependency problem can be solved and you can install librealsense2-utils on your ubuntu 14.04.